forkineye/ESPixelStick #1: Add Grouping Support

forkineye, 2015-09-09:
Add the ability to multiply pixel inputs for "grouped" outputs.  For example, "group by 10" would generate 10 pixel outputs for every pixel input (3 channels) received via E1.31.

madsci1016, 2016-01-12:
Have you done anything towards this yet? It's been bouncing around my head for a little while. Especially a good way to present the configuration on the web page.

forkineye, 2016-01-12:
I hadn't yet. I was thinking of a single "Grouping" entry, with either a text box or drop down for group size (defaulted to 1).

forkineye, 2016-12-13:
Add reversing as well

marx1, 2018-10-11:
I have a need for this. Until it's added I still have to use a E6804 in an all pixelstick show.

penfold42, 2019-01-04:
I’ll likely have a need for this shortly

Has anyone done anything yet ?
If not, I’ll do it

forkineye, 2019-01-05:
I haven't started on it yet.  I'm planning to redo the ringbuffers in ESPAsyncE131 this weekend, but it shouldn't affect this.  Thanks!

penfold42, 2019-01-06:
#147 

Done in pixeldriver for ws2811.
Should be easy to adapt for serial - I have no fixtures to test with

forkineye, 2019-01-06:
ws2811 is fine, thanks!

closed 2019-01-06 by forkineye