forkineye/ESPixelStick #5: Universe spanning causes huge performance drop and doesn't span to next universe

bircoe, 2015-11-28:
I'm attempting to get universe spanning working, I have a spiral tree with 336 WS2812B's, if I set it to Universe 9 (this fits with the rest of my configuration, PPU is set to 170 and pixels is set to 336, the performance drops incredibly, maybe 1-2fps and it doesn't listen to universe 10. If I keep the number of pixels at 170 or below it behaves just fine.

madsci1016, 2015-11-28:
I have 192 pixels running fine on this code. Question, are you trying to run multicast or unicast? Multiple universes is only supported in unicast mode.

bircoe, 2015-11-29:
Thank you, I was using multicast, I wasn't aware that multiple universes don't work with multicast, it is now working well.

closed 2015-11-29 by bircoe