forkineye/ESPixelStick #6: Multicast not responding to IGMP query

madsci1016, 2015-12-06:
I'm having issues using multicast mode through Layer 3 switches, routers and APs. It works for a few minutes, than all nodes stop receiving data at once. Power cycling the ESPs fixes the issues for another few minutes. 

If I rewire the network between my show data source and the AP to use layer 2 switches it works fine. 

I can't confirm this because I don't have anything to capture WIFI traffic to confirm, but i suspect after the timeout my layer 3 switch and router will send out a IGMP query request, get nothing back and then stop all multicast data from transversing. This is probably an issue with the ESP multicast driver, but i wonder if we call udp.beginMulticast() at a set interval we can work around the issue by sending IGMP join requests more than once. 

I first ran into this problem with a simple setup, and realized just my AP was doing IGMP snopping and after not getting any responses to query messages would also stop letting the multicast data through.

madsci1016, 2015-12-12:
Yeah I've done some more testing, and all my Level 3 networking gear blocks multicast after exactly 2:20 after power-up, so seems the ESP does not respond to queries. After Christmas I'm going to try calling the multicast begin line every 2 minutes to work around the issue, and hope it does not introduce noticeable effects with the lights.

forkineye, 2015-12-17:
I have this on my plate to look into, but won't be until the new year. Thanks!

forkineye, 2017-08-10:
Haven't looked at query responses, but finally got around to tracking down the wonky IGMP membership report intervals - https://github.com/esp8266/Arduino/issues/3498.  Fix has been merged into the upstream code for Arduino ESP8266.

forkineye, 2017-08-12:
Pretty sure this isn't an issue anymore. Let me know if it is and will reopen it.

closed 2017-08-12 by forkineye