forkineye/ESPixelStick #7: Erratic random flickering

marcfon, 2015-12-17:
I have a string of 80 leds ws2812b hooked up to an esp-01e that is loaded with ESPixelStick 1.3. I've successfully managed to set everything and can run sequences from xLights (4.2.25) & FPP (1.7). But there is one big problem that I've failed to fix.

With any sequence I run the whole string flickers with random colors in during the sequence. As soon as no more data is send the flickering stops (and the leds keep their last color). So the issue somehow seems related with the processing of data.

One thing that sort of helps is to set the Pixel Count / PPU somewhere around 50 or lower. It doesn't fix it but at least the flickering rarely occurs then.

Also turning Gamma map on/of makes a difference. When Gamma map is on the flickering is much less, it looks like only some pixels flickr. Whilst when Gamma map is off the every pixel seems to flickr.

I've ran out of clues how to debug / fix this issue so I hope anyone here can give me some pointers.

sticilface, 2015-12-17:
What CPU speed have you set 80Mhz or 160Mhz ?

marcfon, 2015-12-17:
I've set the cpu to 80Mhz.

forkineye, 2015-12-17:
Are you using the ESPixelStick boards? If not, what's your connection like between the ESP-01 and first pixel? Any line buffers or resistors? Also, how long is the connection?  As for your wireless network, are your E1.31 sources (xLights and FPP) hard-wired or wireless?  They need to be wired, and make sure only one is transmitting at time (I've been bitten by this many times with Vixen running idle).  Are you using multicast or unicast?

forkineye, 2015-12-17:
Good call @sticilface. You'll want to set the CPU to 160MHz as well.

sticilface, 2015-12-17:
Nae worries. Like to help when i can!  not often... but sometimes!

marcfon, 2015-12-17:
Are you using the ESPixelStick boards? If not, what's your connection like between the ESP-01 and first pixel? Any line buffers or resistors? Also, how long is the connection? 
- I'm not using the ESPixelStick board. Not sure I understand your question but the dataline of the ws2812b is directly connected to GPIO2 of the esp01e. No buffers or resistors. I've controlled the strip before using the same setup but different software and never had any issues with flickering.

As for your wireless network, are your E1.31 sources (xLights and FPP) hard-wired or wireless? 
- FPP is hardwired and xLights is wireless.

The strange this is like I said that when the transmission of data stops the string can perfectly well hold it's color without flicker.

I'll see if it helps when I flash with cpu set to 160Mhz.

Thanks for the lightning fast support!

marcfon, 2015-12-17:
@sticilface 160Mhz totally did the trick! Thank you. 

I'm just 2 days into this so it's like seeing magic :)

closed 2016-06-09 by forkineye