forkineye/ESPixelStick #8: Add TLS 3001 support

forkineye, 2016-01-08:
http://doityourselfchristmas.com/forums/showthread.php?39166-ESPixelStick-E1-31-WiFi-Pixel-Controller&p=413124#post413124

forkineye, 2016-09-28:
Protocol references:
https://github.com/FastLED/FastLED/issues/1
http://auschristmaslighting.com/forums/index.php?topic=1938.0

MartinMueller2003, 2020-11-07:
Just looked into this.  The current unify ws2811 output driver with a modified ISR should be able to do this. I would need two or three TLS3001 based pixels to test with.

MartinMueller2003, 2020-12-06:
These pixels are much more expensive than the WS2811 devices. If you send me a few, I will take a crack at setting up the protocol. Assuming you still want to add this to the Unify branch.

forkineye, 2020-12-07:
Going to close this out. Don't know of anyone that uses them anymore.

closed 2020-12-07 by forkineye