forkineye/ESPixelStick #14: Add DMX 512 Support

Blickensderfer, 2016-06-30:
Having DMX output option would be awesome.

forkineye, 2016-06-30:
It's in there, but I haven't been able to test it yet - https://github.com/forkineye/ESPixelStick/commit/15ffe33d532523bfd99688914f3359fa7afc0c6e

Blickensderfer, 2016-06-30:
I would like to test it if you need testers.

forkineye, 2016-07-09:
Verified DMX is working.  You'll need to either use a solution that provides RS485 output, or add a RS485 transceiver to the ESPixelStick hardware.

closed 2016-07-09 by forkineye

Blickensderfer, 2016-07-11:
I would use the Data and connect it to the Data In on the RS485 circuit something like this?
[image]

forkineye, 2016-07-11:
@Blickensderfer That should work.  You can go that route, or just pull the rs485 chip in your existing controller and feed the data line straight into the host processor, similar to how Bill does with his - http://www.billporter.info/2015/09/05/renardesp-wifi-adapter-for-renard-dimmers.  The data line and ground should be all you need.

petiepooo, 2018-04-08:
Closed issue, I know, but I'm wondering if a RS422 transceiver like the 8-lead no-enables chip shown at https://www.intersil.com/content/dam/Intersil/whitepapers/interface/rs-485-transceiver-tutorial.pdf would work.  That would avoid the need for the passive on the RO pin.