forkineye/ESPixelStick #16: OTA Firmware update howto?

marcfon, 2016-07-30:
Currently I'm running the latest master branch. I see the option to update the firmware OTA. Now I'm just wondering which file I should upload as the data/www files have to be uploaded separately to SPIFFS.

forkineye, 2016-07-30:
It's not fully implemented yet.  Right now, it only handles the compiled sketch binary, similar to the standard Arduino ESP8266 OTA stuff.  I have a tool planned that'll combine the sketch and SPIFFS images into a single file for upload, but will be a few weeks before I can start on it.

marcfon, 2016-07-30:
I understand. So as long as the code changes aren't related to data/www the firmware update should work just fine right?

forkineye, 2016-07-30:
Yep, as long as you aren't changing the web handler interface stuff you'll be fine.

marcfon, 2016-07-31:
Great! Not really a question about the firmware but if I would want to do light up some pixels on startup as a test sequence, would that be possible from within the current code? I guess I would need to create a data packet and push that to the pixels right?

marcfon, 2016-08-02:
Where could I look for a clue on how to put the right data in the buffer? Let's say I'd want to light the first led green or something like that. An alternative approach that I'm thinking of is just to use the adafruit neopixel library to achieve this... or would that be silly?

forkineye, 2016-08-02:
Inside the main loop where is sets the value:  https://github.com/forkineye/ESPixelStick/blob/master/ESPixelStick.ino#L535

Just call the following, it's in RGB order - pixels.setValue(address, value);
To light up max green for the first pixel - pixels.setValue(1, 255);
To light up max red for the second pixel - pixels.setValue(3, 255);

marcfon, 2016-08-04:
Works just as advertised! Easy enough to light up pixels (channels). Thanks.

forkineye, 2016-09-28:
OTA updates via web page implemented - https://github.com/forkineye/ESPixelStick/commit/846858fa59045190a16f6da1cd19c3d0e7cce8fa

closed 2016-09-28 by forkineye