forkineye/ESPixelStick #17: WS2812 DMA Control?

nrudolph7, 2016-08-14:
Hello! Thank you for your code! I have been using it on some ESP8266 breakout boards with great success!

I recently tried using a D1 Mini board (esp8266) that seems to have trouble with the bitbanging that is used to control the pixel data line. Is it possible to rework the code to control the pixels with either DMA or UART? My problem is that the pixels will light the assigned color, but they will stay on for about 2-3 seconds and then flash off for about a second, and the cycle repeats. I believe this is because of the bitbanging on gpio2, which also has the blue led of the esp8266 module? Maybe you have some other insight as to what is causing this issue? 

Here is a thread detailing some work done on WS2812 leds using DMA on the ESP8266 (and D1 Mini) boards, as well as a link to the Makuna NeoPixelBus Library that seems more up to date for current ESP8266 boards.

https://github.com/Makuna/NeoPixelBus/issues/88

https://github.com/Makuna/NeoPixelBus

If it's not possible to incorporate a DMA option, perhaps it is possible to change the pin that the bitbanging occurs on?

forkineye, 2016-08-14:
Its not big-banged, it's already delivered via UART asynchronously.  The original UART implementation for that project actually came from the pixel code in this one, then they added the DMA and async support which I later rolled back into this one (async).  It sounds like something else is going on.  What are you using for your e1.31 source?  Is it wireless as well? If so, it needs to be wired.  You'll get all sorts of weird lag issues and hiccups since 802.11 is simplex.

nrudolph7, 2016-08-14:
Thank you for such a quick reply! I saw several references for bitbanging in your code and just incorrectly assumed that's what it was. Thank you for clarifying and helping me out! I'm using Chamsys MagicQ PC to deliver E1.31 DMX signals over WiFi. What's strange is that your code works perfectly with Chamsys over WiFi on an Adafruit Feather ESP8266 Huzzah, but not the D1 Mini. I have the data line connected to GPIO 2 (pin D4 on the D1 Mini board), and the pixels flicker out on a regular cycle. If I disconnect the data line, the pixels will stay lit, so it's definitely something happening on the data line.

I should be using GPIO 2 for data, correct?

Here is the board I am trying to use, with issues:

http://www.wemos.cc/Products/d1_mini.html

Here is the board that works perfectly:

https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/pinouts

Do you have any suggestions for what could be causing the flickering? It's almost like the signal timing or cycle is being disrupted. They don't flicker, but rather just shut off for a second and turn right back on, in a regular pattern.

nrudolph7, 2016-08-14:
Fasle Alarm! It's working perfectly now... I'm still using all WiFi, but I'm on another router now. Very strange, as the previous router worked fine with all WiFi connections, though I wasn't streaming netflix to a Roku over ethernet. Maybe that was the culprit? I'll do some testing later tonight.

Another question for you if you don't mind! I'm looking to build 200 circuits consisting of an ESP8266, three ws2812 leds, powered by portable usb battery packs. These will all be installed within about 60 to 70 yards from the router. If I am using multicasting over WiFi, with three leds per board (so, about 3.6 full universes - each board would be receiving multicast for 9 unique channels), will the WiFi connections be able to hold up?

The router will not be connected to the internet or used for anything other than multicasting the E1.31 signal. Maybe I'll need several routers and bridge them together to distribute the signal? I don't really know how efficient, fast, or reliable E1.31 and ESP8266 is in a large scale multicast setup like this.

forkineye, 2016-08-15:
@nrudolph7 The only problem in that scenario is having 200 units associating with your AP. Most AP's will start to crap out way below that.  You'll need a couple of enterprise class AP's and a well thought out band plan / site survey to pull it off.  If it was me, I'd dump WiFi all together and go with a nRF24L01 based solution in that scenario.  The overhead of 802.11 (link layer handshakes, station tracking, etc..) is what's going to be the big hurdle.  It'll easily handle a few thousand channels, just not near that many WiFi nodes. With a nRF24L01, you can just blast it out and each node picks what it wants.  The predecessor to this project was such a solution and I have it documented here along with a link to the code if interested - http://forkineye.com/ws2811-easy-way-dma-xcl-xmega8e5/

closed 2016-09-28 by forkineye