forkineye/ESPixelStick #18: Support more than 170 LEDs with a raw protocol ?

penfold42, 2016-08-19:
Hi there - great work on this project. It's so good I'm
Basically I'm about to kill my own and extend this instead

I've just added e1.31 support to https://github.com/hyperion-project/hyperion.ng/pulse and it works great.

Some Hyperion users need >170 LEDs so I need to look at some options and I'm curious as to your thoughts on some approaches
1) we already have a raw protocol which is good for about 500 LEDs (limited by no UDP reassembly support in the ESP). It should be easy to add another listener.
2) use multiple universes ? Could we  map universe 1 to the 1st 170 LEDs, universe 2 to the next 170 LEDs and so on?
3) violate the DMX protocol and simply send more LEDs ?

Thoughts on preferred approach both in general and what you'd consider merging ?

Thanks again !

forkineye, 2016-08-19:
I would just roll into the next universe and send another packet (option 2).  Note though the ESPixelStick code uses all 512 channels of the sACN packet, so it wouldn't be aligned to 170 pixels per universe.  With 171 pixels, RG would be the last two channels of the first universe, with B being the first channel of the next.  Think of it more as one continuous data stream and the universes just segmenting that stream.  You will however want your e1.31 source (hyperion I'm guessing?) to be wired though due to the simplex nature of 802.11.  It works with lower channel counts, but can cause bad lag issues once the channel count goes up.   Here's the ESPixelStick driving 720 pixels - https://www.youtube.com/watch?v=perR5NHttQ4

penfold42, 2016-08-20:
Perfect - thanks for your detailed reply.

I didn't realize it magically rolled over

closed 2016-08-20 by penfold42