forkineye/ESPixelStick #19: Your interest in other features ?

penfold42, 2016-08-20:
I was working on a similar project but have decided to ditch it and base it on yours.

Below is what I have done or wanted to do that yours doesn't. Are you interested in PRs or violently against:
1) UDP raw protocol support - listen on another port which just has unframed raw RGB data. (done)
2) a UDP ascii interface to control and query status (done, but mostly useful for testing)
3) web interface to view current led colours (cute but more diagnostic than useful)
4) simple web UI to directly manipulate the colour of the strip (not started yet)

Thanks !

madsci1016, 2016-08-20:
For the target audience of this project, these are unusual features that will go vastly unused in my opinion. The precedent set by other pixel controllers consists of industry protocols and web based configuration management.

forkineye, 2016-08-23:
I'm not too keen on adding support for non-standard protocols right now.  I've had a few requests for MQTT support, but need to investigate it further.  As for a control and query interface, there will be a RESTful type configuration interface and web based testing interface.  Once I wrap up 2.0 with OTA support, the entire front-end will be changing so I'd hold off on doing much with the current front-end.

penfold42, 2016-08-23:
Thanks - I was playing over the weekend and got a basic javascript colour picker http posting to set the whole strip to a single colour

closed 2017-08-12 by forkineye