forkineye/ESPixelStick #21: Add ArtNET support

mdorenkamp, 2016-09-11:
It would be nice to have ArtNET support. I already have code that acts as ArtNET node and outputs to serial if that helps.

Jearde, 2016-09-11:
I totally agree! I also use a build with ArtNet support and an OLED for knowing, whats configured. These two things are very important for me to have an easy to use setup. I already have a Fork on the old version with OLED and ArtNet support, but I do not have the time to update it to the master branch. If you are interested, the code is in here:
https://github.com/Jearde/ESPixelStick

MartinMueller2003, 2020-11-22:
Why would we go to an older protocol? We have E1.31 and now DDP. Does ArtNet solve any problem that cant be solved using one of the supported protocols?

forkineye, 2020-11-22:
It still seems to be popular in certain areas and countries. I think mostly due to the newer protocols not being supported by the software they're using. I have a library I was writing for it somewhere, I'll dig around my archives.

MartinMueller2003, 2020-11-22:
If you get me the lib, I will integrate it.

jnovack, 2020-12-30:
https://github.com/hideakitai/ArtNet

forkineye, 2021-09-24:
In main and staged for 4.0 release.

closed 2021-09-24 by forkineye