forkineye/ESPixelStick #30: Assign DMX channel to GPIO

starwolf73, 2017-01-05:
Hi,
Sorry if this is not the place, I didn't find your forum to post this request. 

Thanks again for sharing this great software. Our Xmas tree was magic this year as I used your software on a Wemos D1 mini with XLight sequencer. Work perfectly. 

I was wondering if it is possible to use other available GPIO on the ESP8266 and assign it to a universe/DMX Channel. My goal is add a relay or dimmer (if PWM possible) to control other normal light. 

Thanks
Eric

forkineye, 2017-01-08:
It should be fairly easy to do.  I have another project in the planning stages that will require software PWM, just no time to work on it yet.

starwolf73, 2017-01-10:
Great! Thanks. Looking forward to test it.

penfold42, 2017-08-05:
I have some hardware arriving soon which could really use this feature.

Did you get anywhere ? If not I'll probably do it in the next couple of weeks.

I'll add a gpio/PWM tab to the GUI with 3 columns:
Gpio, enabled, dmx channel
12 [x] 1
4 [x] 2

forkineye, 2017-08-05:
@penfold42 I hadn't done anything with this yet.  If you decide to work on it, I'd like to keep it as a different compile time "Mode", like Pixel and Serial as the hardware most people use this on (ESPixelStick) wouldn't allow for mixed capability due to lack of available pins (ESP-01).

penfold42, 2017-08-05:
Will do - I just got the basics working 5 mins ago.

Ideally I'd like to keep it as separated as possible but the globals being in the same .h with all the structure makes it trickier - a subtle hint to peek at my other PR !

penfold42, 2017-08-07:
https://github.com/penfold42/ESPixelStick/tree/pwm?files=1

It's basically done - the GUI layout looks crap at the moment but it works.

I've only tested gpio4 with a mosfet driving a 5w 12v dimmable led halogen replacement bulb.

penfold42, 2017-08-18:
I've fixed the layout and confirmed support for most of the gpios. 

I just need to hide the HTML config if support isn't compiled in.

Do you mind if I change device mode to be a bitfield? Pwm can coexist with both serial and pixel modes and this is a quick way to support it.
Bit 0 pixel, bit 1 serial, bit 2 pwm
Would yield:
0 invalid
1 pixel
2 serial
3 invalid
4 invalid
5 pixel + pwm
6 serial + pwm

penfold42, 2017-08-21:
#66

penfold42, 2018-07-12:
And now removed from this repo as of today.

I’ll be maintaining this support in my repo along with other features:
https://github.com/penfold42/ESPixelBoard/tree/ESPixelBoard?files=1

closed 2018-07-12 by forkineye