forkineye/ESPixelStick #31: Page not found

asm7100, 2017-02-19:
Hi just trying to program a Wemos D1 mini, with your firmware.
All goes good.
But when i try to connect my browser to it it says "page not found"
when i try "192.168.0.199/conf" i get this 
{"device":{"id":"ESPixelStick","mode":0},"network":{"ssid":"Aaron","ip":[192,168,1,10],"netmask":[255,255,255,0],"gateway":[192,168,1,1],"dhcp":true,"ap_fallback":false},"e131":{"universe":1,"channel_start":1,"channel_count":510,"multicast":false},"pixel":{"type":0,"color":0,"gamma":true}}
Its like is not have the html files and that upload says all good?

cinoan, 2017-02-19:
I have also set up a wemos d1 mini this weekend and it has worked ok. Did you follow the instructions in https://github.com/forkineye/ESPixelStick/blob/master/html/README.md to compress the web files?

asm7100, 2017-02-19:
No iam running on a windows system :-(

asm7100, 2017-02-19:
I have used your Manual file.
http://forkineye.com/wp-content/uploads/2016/09/ESPixelStick-V2-Assembly-and-Usage.pdf

asm7100, 2017-02-19:
This is the output from the compiler.
But still "Page not found"
 
Sketch uses 316879 bytes (30%) of program storage space. Maximum is 1044464 bytes.
Global variables use 39376 bytes (48%) of dynamic memory, leaving 42544 bytes for local variables. Maximum is 81920 bytes.
Uploading 321024 bytes from C:\Users\Allan\AppData\Local\Temp\arduino_build_926680/ESPixelStick.ino.bin to flash at 0x00000000
................................................................................ [ 25% ]
................................................................................ [ 50% ]
................................................................................ [ 76% ]
..........................................................................       [ 100% ]
[SPIFFS] data   : C:\Projekter\DMX\ESP8266\ESPixelStick\data
[SPIFFS] size   : 1004
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/config.json

[SPIFFS] upload : C:\Users\Allan\AppData\Local\Temp\arduino_build_926680/ESPixelStick.spiffs.bin
[SPIFFS] address: 0x300000
[SPIFFS] reset  : nodemcu
[SPIFFS] port   : COM3
[SPIFFS] speed  : 921600

Uploading 1028096 bytes from C:\Users\Allan\AppData\Local\Temp\arduino_build_926680/ESPixelStick.spiffs.bin to flash at 0x00300000

................................................................................ [  7% ]

................................................................................ [ 15% ]

................................................................................ [ 23% ]

................................................................................ [ 31% ]

................................................................................ [ 39% ]

................................................................................ [ 47% ]

................................................................................ [ 55% ]

................................................................................ [ 63% ]

................................................................................ [ 71% ]

................................................................................ [ 79% ]

................................................................................ [ 87% ]

................................................................................ [ 95% ]

............................................                                     [ 100% ]

forkineye, 2017-02-21:
If compiling from source, the html files have to be processed with Gulp -https://github.com/forkineye/ESPixelStick/blob/master/html/README.md.  It can all be done on a Windows system.

closed 2017-06-25 by forkineye

DJSGXII, 2017-09-14:
How to upload it with Windows? Please help me understand.

forkineye, 2017-09-15:
@DJSGXII Recommend you download the latest RC from the release tab which contains pre-compiled binaries along with a flash tool and follow the README included in the release package.

DJSGXII, 2017-09-15:
@forkineye Thanks for the quick reply. I got it working now. You are the best!

priyankchouksey, 2018-10-01:
> @forkineye Thanks for the quick reply. I got it working now. You are the best!

How did you get it working? I am not able to upload using the jar, always getting esp sync issue. I tried building and upload the ino from source, added the precompiled data/www folder from the package to the ino/data folder and modified the json file with SSID/passphrase and ip address. After all this uploaded SPIFFS. All done well but the ESP does not connect to Wifi and I am not able to access it.

Any help/Steps would be great help.