forkineye/ESPixelStick #41: Multicast change not implemented until next reboot

cinoan, 2017-03-08:
If you enable/disable multicast on the Device Setup tab and save changes the change isn't implemented until the next reboot.

Options.
1) Move Enable Multicast to Wireless Setup tab which forces reboot on change.
2) Detect change of setting and force reboot (need to tell the client that it is now waiting on a reboot rather than an update)
3) Force Multicast enabled it still responds to unicast in that status.

I think option 1 is the best unless you think 3 wouldn't cause any issues. I'm happy to implement any once it's decided.

forkineye, 2017-03-08:
I had originally had it on the wireless tab and had moved it.  Reason being is that I wanted to keep anything DMX specific out of the wireless setup tab in case I add other listeners later (like ArtNet).  We should be able to handle multicast configuration without rebooting, but not with how the code is currently structured (coupled to the wireless setup).

forkineye, 2017-08-04:
Fixed as of https://github.com/forkineye/ESPixelStick/tree/v3.0rc2.

closed 2017-08-04 by forkineye