forkineye/ESPixelStick #42: Testing status not displayed on reconnection

cinoan, 2017-03-08:
If you enable a testing mode, e.g. Rainbow, then reload the interface it looks like testing mode is disabled but the output isn't responding to incoming data. 

Options.
1. Update interface to show current testing status (would require a way for the client to get current test settings, e.g. colours, column counts, etc)
2. Disable testing mode on a new connection. As current testing status is sent to client on load a notification could be shown that testing is disabled.

I think on balance Option 2 might be the best choice. It would be simpler than Option 1 and I think an unexpected exit from test mode is better than unexpectedly being in test mode. Again I'm happy to do this once it's decided on the best route.

forkineye, 2017-03-08:
My intention was to implement this as option 1, but just haven't had time to do it.  This of course was before you all implemented the other test modes and I haven't had a chance to really look at them yet.  The idea however, is that "Test Mode" could also double as off-season lighting control, allowing people to use their ESPixelStick for accent / mood lighting, etc...

cinoan, 2017-03-10:
I've been thinking about this a bit and you could end up adding more and more features to the test section as people want to do different things - and I'm sure you don't want to recreate xlights!

Would it be feasible to run with 512KB SPIFFS and allow the user to upload a mini sequence to play?

Currently just over 57KB bytes of filesystem space is used, of which about 50KB is libraries - Bootstrap, jquery & the colour picker. If you assume non library code grows less than 4 times (to 28KB) that leaves 434KB that could hold a sequence. With 200 pixels at 20fps that would be about 37 seconds of sequence. Not great but possibly more flexible than lots of test modes.

I'm not sure how much space you need for the OTA updates and if that limits you to smaller SPIFFS sizes.

I was going to say you could have a version with SD cards for longer sequences, or just have one ESP with an SD to act as a player to normal sticks, but the other consideration is how many people would use ESPixelSticks and not be happy to have FPP running to control their off season use.

madsci1016, 2017-03-11:
I'd argue local playback of mini-sequences is outside the scope of firmware used for streaming playback in a device used in a larger system, but would be cool for a permanent fork of this project or the creation of another project. The you wouldn't have to be worried about killing OTA updates. 

Some ESP devices come with 2 or 4 MB which would be even more space. I could see something like the Java flasher tool used to upload a SIFFS image with a FSEQ file and the firmware for local playback as small as possible. Would be nice to have NTP for triggering playback per a schedule, and a simple web user interface,