forkineye/ESPixelStick #45: Add SK6812 RGBW Support

forkineye, 2017-05-03:


penfold42, 2017-07-10:
Is there value in supporting both a 4 channel per led mode (direct RGBW) and. 3  dmx channel mode where we just set white to off ?

BioSehnsucht, 2017-09-06:
I bought two 5m / 150 pixel strands of sk6812 about a year ago and never got around to putting them to use. I'm interested in this functionality. Looks like I might need 4 or more ESPixelStick (at 0.3W/pixel) to stay under 5A each, while I don't have a lot of time to dedicate to this right this moment I'd be willing to help test by wiring up a length of SK6812 to an ESPixelStick and trying different builds.

Ideally I'd like to see RGBW implemented (as having a W component was why I bought these particular strands), but implementing either RGB with W off or some kind of "smart" mode that sets W according to the common value of RGB (and subtracts that out from RGB) might be useful too, to emulate RGB functionality in situations where the software outputting the sACN stream can't do RGBW or adjusting it to do so would be too much work. (though I'm not sure how much horsepower is available to the ESP, perhaps doing all that math per-pixel per update is too much to ask for?)

If I had more free time I'd take a stab at implementing it myself but I'm tapped out at the moment / near future.

penfold42, 2017-10-26:
I’ll do this next on my list.
I plan to extend the colour order options to include:
- RGBW White off
- RGBW Sutract Min
- RGBW pass thru (assumes 4 bytes per led in the incoming network packet)

nwaxiomatic, 2018-01-22:
any updates on this?

penfold42, 2018-01-22:
I started but it crashes

I suspect I’m over running a buffer somewhere as I expand the data by 4/3

penfold42, 2018-01-22:
https://github.com/penfold42/ESPixelStick/tree/sk6812rgbw-support?files=1

starwolf73, 2018-05-05:
I'm also very interest and willing to support if I can. This will definitely be a new ground when this will be achieve and open up door to great application.

Moonbase59, 2018-08-09:
What would you all expect? Mainly a "kind-of-RGB" fixture that auto-mixes the white component or a real 4-channel/pixel fixture?

Calculating the white part out of RGB might need some extra tables, since the strips can be had with cold/neutral/warm white and the white LED is actually very bright, so might need to be restricted in brightness. We also might need an extra 4-channel gamma table (?).

Then again, I’d also love to have it and even buy some strips and help, if possible.

Fortunately, the SK6812’s timing is quite near the timing of the WS2812B, so we just _might_ get around by defining an extra "white" byte, introduce a CRGBW struct and a larger pixel buffer.

I still wonder 
* if serial Renard supports 4-channel mode, and
* if we might need a different color order table (RGB, GRB, … but also including white).

I seem to remember that [FastLED](https://github.com/FastLED/FastLED) supports SK6812 but they did apparently not introduce a new CRGBW struct. Might have to look up the code.

nwaxiomatic, 2018-08-09:
For my setups I've been running a hard 32-bit RGBW int line to the lights and controlling the "whiteness" from the controller. I think allowing the struct to handle this hard code would be best and leave the calculation preference to the user.

penfold42, 2018-10-19:
Cool, thanks.

I’ll look into finishing this and getting a PR submitted.

You’re right about supporting both options - I think we actually need 3 options:
- incoming RGB (white set to 0)
- incoming RGB (white calculated)
- incoming RGBW (passed thru)

NormenL, 2019-03-29:
Not sure if anyone has written about this but, just got the SK6812 RGBW working on the Enttec Led Mapper. In ESP settings I changed it to RGB and in ELM I it to GRBW

matajumotors, 2019-07-30:
Hi guys, have you done any progress on converting rgb input into rgbw pixel output on the pixelstick? Thank you.

dennispg, 2019-12-06:
I took a stab at implementing this and want to share it here for anyone else interested in it:
https://github.com/dennispg/ESPixelStick/tree/sk6812rgbw

For now, it basically just ignores the W pixel but it works to drive an SK6812 strip as if it were RGB.

I'm also still working on extending the effects tab in the web admin portal to allow for independent W control.

I don't think I know enough yet about e1.31 or any of that to get into the weeds of allowing for incoming RGBW, but once I finish indepedent W control from the web portal this will serve my needs perfectly.

forkineye, 2019-12-06:
This should be much easier to add once the unify branch is done which implements a modular system for input and output code. I wouldn't work too much harder on it until 4.0 which will merge the unify branch into the master.

dennispg, 2019-12-06:
I added support for controlling the white channel from the web portal, and went ahead and submitted a PR with what I've done.

You don't have to merge it if it interferes with 4.0 related work. Just thought it would make it easy to find for anyone to whom this might be useful to have the PR there at least.

forkineye, 2020-11-06:
Support is in the Unify branch now. Appreciate anyone who could test it and give feedback.  Thanks!

patdelaney, 2020-11-24:
I ordered the pixels to test this.

bkmcleod, 2020-12-04:
I tested this tonight on a 10pixel long strip of RGBW. No issues found, other than difficulty configuring within Vixen 3.0, as it does not support 4 channel with color mixing to my knowledge.

patdelaney, 2020-12-04:
My Pixels arrived I will try to test. @bkmcleod I believe that xLights supports RGBW  you might try that.  That's what I'll be testing.

patdelaney, 2020-12-04:
@forkineye I don't see a dropdown for SK6812  am I missing something?  I pulled in unify branch.

forkineye, 2020-12-05:
> I tested this tonight on a 10pixel long strip of RGBW. No issues found, other than difficulty configuring within Vixen 3.0, as it does not support 4 channel with color mixing to my knowledge.

Thanks!

> @forkineye I don't see a dropdown for SK6812 am I missing something? I pulled in unify branch.

I haven't had a chance to look at it yet. Will try to check it out in the morning.

forkineye, 2020-12-06:
> @forkineye I don't see a dropdown for SK6812 am I missing something? I pulled in unify branch.

It's under ws2811 as RGBW.

patdelaney, 2020-12-06:
Okie doke  I'll give that a try.. Have any of the "test" modes in the Pixel stick been updated to specifically test RGBW strips? Where W is W?

forkineye, 2020-12-17:
Should we leave this under WS2811 as RGBW or break it out into a SK6812 module?

MartinMueller2003, 2020-12-17:
The data protocol is the same. I vote for keeping it as RGBW :)
Martin

forkineye, 2020-12-21:
Supported in Unify as WS2811 with RGBW color order.

closed 2020-12-21 by forkineye

UbhiTS, 2022-10-18:
Can someone please help me.

I have SK6812 RGBW led strips controlled via an ESP32 running WLED.
When I go into xlights > layout > model > controller connection, I do not see a dropdown for the sk6812 protocol. I've tried to use ws2811 with different permutations and combinations of Color Order as well as String Properties but to no success. The moment I upload the output to the controller, it changes the LED settings on the WLED. Since SK6812 is not selected in WLED, it makes my controller unusable.

1. Can you please add the SK6812 native support to xlights
2. Can you please give me the settings to run SK6812 led string on a WLED controller with xlights

Thanks

UbhiTS, 2022-10-18:
okay, I figured it out, there were multiple settings I needed to fix in both WLED and xLights. If anyone is stuck in the same situation, I hope this helps :)

[image]

MartinMueller2003, 2022-10-18:
FYI: This is the ESPixelStick repository. You would have needed to post your question to an xLights or WLED forum to get an answer to your issue. There is no way we could have fixed this for you. FYI: ESPixelStick V4 has the ability to be built with SK6812 RGBW support.

UbhiTS, 2022-10-18:
oops, sorry :) ... my bad I just read "add sk6812 support" and then I went on my keyboard rampage :) ... my apologies. should I delete my comments?