forkineye/ESPixelStick #48: ws2801 support?

blackrockcitynv, 2017-05-08:
was this ever added? will it ever be?

forkineye, 2017-07-10:
I had intentions of adding this when starting the project, however the ESP-01 does not breakout the pins required to do it properly.  I have plans to added it in the future for ESP-12 based modules, and possibly modification of the ESP-01 module for current ESPixelStick hardware.

bquesse, 2017-07-10:
This would be great. A good many of us roll our own pixelsticks with nodeMCU modules (esp-12e), so this would be a useful addition.

penfold42, 2017-07-20:
The uart TX would be a safe "spare" pin for a software SPI clock line if we avoided serial.print.

I've used the https://github.com/JoaoLopesF/ESP8266-RemoteDebug-Telnet library in another project with minimal rework

forkineye, 2017-07-20:
I have simple implementation that seems to work using HSPI on a ESP-12.  I tested with MOSI and TX tied together as well and seems to work fine.  It needs to be optimized, but seems it'll work.  Will push a feature branch for it in a bit.

forkineye, 2017-07-20:
Initial implementation, still needs optimization - https://github.com/forkineye/ESPixelStick/tree/feature-ws2801

thommy86, 2017-09-02:
The Github version of the feature-ws2801 wasn't able to compile, so I merged it with the master. This is working as expected now. But as I am using APA102 I was wondering if you are going to add support for this. The leds are full white now and I am not able to control them.

kenmcc, 2019-11-28:
Is this merged to master, or  still out on the branch? which pins are used for WS2801 please?

bquesse, 2019-11-29:
Ken, I merged that branch for my show in 2017, so it's a fair bit behind the master now, but I still have it functioning in my show for the couple hundred pixels that I still own. I could share if you'd like.

kenmcc, 2019-11-29:
Hi, that would be great, if you could, thanks.
though I suspect I'll run into the same issue I had last night - all the libraries (E131ASync etc) have moved on since then, also ArduinoJson has moved on so all the Json stuff has changed from the version on the branch.

I managed to get it pretty much working last night, by building master, and then copying the WS2801 stuff (as much as I could find anyway) from this branch to what seemed to be the right place in master. And it mostly works, I think on D5 and D7 pins on my nodemcu, but it's sluggish - at times when i set the string (50) to say red, one or 2 in the middle will stay the previous colour for a couple of seconds before settling to red.

I'm brand new to xmas lighting stuff, so I don't yet know if the problem lies in my ws2801 espixelstick code, xLights output (am using xLights directly to drive the pixelstick on my home network, rather than e.g. fpp, while i get up to speed with it), or my WiFi, though that's fine for streaming video so I can't see how that would be the cause.

I'm awaiting WS2811 leds from aliexpress, so when they come and i get them running alongside, i should be able to see where the issue really lies. I've had the ws2801s for about  6years driven from an arduino nano - nothing fancy, just cycling through some predefined effects and colors, swapping on button press input.

But if you could share yours anyway, I might spot a discrepancy I made with my code last night.
Many thanks

bquesse, 2019-11-29:
Sure thing -- I remember doing much the same when a merged the code in 2017. I believe that the libraries you speak of are also locked in time on my computer, so I'll include those too. One thing I've found is that even moreso than the usual ESPixelStick code, the WS2801 branch does not like to be serving up the web interface and outputting to the lights at the same time, so make sure you close your windows after you check in on it. I'm just heading to sleep now, I'll package up the files in the morning for you. Ping me if it's been more than 12 hours in case I've forgotten.

kenmcc, 2019-11-29:
Oh that'd be excellent, many thanks.  I'll try again this evening (I'm in Ireland!) with the webui shut off and see if that makes any difference. tbh, i just want to use these 2801s to figure out the spacing for a matrix i want to build with the 2811s when they come, so it's not the end of the world if I don't get them running brilliantly (though since I have them pretty much working anyway it would be nice to use them if i can)

bquesse, 2019-11-29:
Here you go! No guarantees that this works, but it's what I have in my files. Good luck!
https://drive.google.com/open?id=1sLMgoSN2jQWq7_kO1l0mxDXekVtttHun

kenmcc, 2019-11-30:
super, thanks a million, i'll have a play with that tomorrow when I get a chance.
much obliged

kenmcc, 2019-11-30:
built it and deployed it - same issue with the tardy sequence from xlights. when running the sequence from fpp it's nothing like i programmed.
Will wait until my 2811s arrive and use the master branch and see what the story is then. Been a fun exercise though, at least I've figured out how to get fpp up on a raspberry pi, and get a sequence over from xlights to it and start it doing something on the espixelstick.

I think i'll also set up a dedicated network.
Cheers for the files!

forkineye, 2019-12-02:
Sorry, I decided to kill WS2801 support a few weeks ago. The v3.0 hardware will include a SD card reader and the SPI bus is needed for that instead.  The branch you all are playing with was just a proof of concept and was never optimized for show usage.  It may be possible to add an output module to the unify branch once it's complete, but it would conflict with the SD support.

kenmcc, 2019-12-04:
no bother, thanks for the work anyway! like i said at least this branch got me up and running with FPP and ESPixelStick while waiting for the boat from China to bring me my WS2811s for really getting going

kenmcc, 2019-12-13:
WS2811s arrived and working as expected, so all issues encountered down to WS2801 interfacing to lights. All good now anyway

closed 2020-02-04 by forkineye