forkineye/ESPixelStick #53: seaprate view stream out from the testing effects ?

penfold42, 2017-07-05:
Maybe a checkbox to enable it and leave the dropdown to focus on running effects ?

What do you think ?

forkineye, 2017-07-10:
I have two thoughts on this.. 1) Just leave it at the bottom of the list as different test modes are added to lessen UI clutter..  2) Separate out stand-alone effects and diagnostic stuff.  I do plan on adding an internal test pattern / effect list that can be published via MQTT so that there are other options besides just static when using MQTT with home automation stuff.  Just not sure how I want to do it yet, but it will be synced with the effects from the test drop-down.

penfold42, 2017-11-18:
Thinking about this again - my vote would be to separate it to a discrete section on the navbar.

If the effects become more sophisticated, configurable or support sdcard playback this becomes even more useful

Kamal-Sonani, 2017-11-18:
@penfold42 thats what i want to do 😎

penfold42, 2017-11-18:
im looking at it now

penfold42, 2017-11-18:
done, #92

closed 2018-07-23 by forkineye