forkineye/ESPixelStick #55: Can't change SSID and Password from Web UI

patdelaney, 2017-07-09:
I am on the new v3.0 beta. I upgraded from 2.0 with no problems. I am new to Pixelstick. I tried to switch the SSID and password from the Web UI, but it will not allow me to save it. Is that a bug or as designed?

penfold42, 2017-07-10:
A bit of both - if you edit the host name it triggers a field validation that enables the save button

forkineye, 2017-07-10:
@patdelaney By not allowed to save, do you mean the button being disabled?  The page now does hostname validation, so if you have a hostname with invalid characters the save button will be disabled.

penfold42, 2017-07-10:
There's a bug here - the button is disabled by default even if a valid host name is already present.

If I just want to change the SSID, I still need to make a tiny edit to the host name to enable the button

forkineye, 2017-07-10:
@penfold42 Ah, thanks!  I'll fix it next time I get a chance to work on the code

penfold42, 2017-07-10:
I can probably do it - I'll add validation for ssid and key too

penfold42, 2017-07-13:
I can confirm #58 fixes this

forkineye, 2017-07-13:
@penfold42  Thanks! I'll merge the PR's this weekend if not sooner.

closed 2017-07-15 by forkineye