forkineye/ESPixelStick #62: sACN Start Code Flashing All LEDs

niloc887, 2017-08-03:
Hello,

I have used your other E131 code and applied the fix that someone has commented on [here](https://github.com/forkineye/E131/issues/3) to stop the sACN start codes from flashing all channels at the priority setting intermittently.

However, the ESPixelStick code is doing the same but I cannot figure out how to ignore the start code packets.

Is someone able to point me in the right direction, or apply a fix directly?

Thanks

forkineye, 2017-08-03:
I'll try to get a fix pushed to the E131 library this evening. Are you able to test if so?  Are you using an ETC console as well?

forkineye, 2017-08-03:
Fixed in https://github.com/forkineye/E131/commit/b6c7071f070ae94d4c69105dbeb126c3fe245920, can you test it?  Thanks.

niloc887, 2017-08-04:
Thanks very much for the quick reply.

I have updated my version and the flickering is now gone.

I am using LSC Clarity on PC. It's a free download but will blackout for 10 seconds every now and again without one of their dongles if you want to test yourself.

[LSC Clarity](http://www.lsclighting.com/download-remository/Clarity/Software/)

forkineye, 2017-08-07:
@niloc887 I'll check it out, thanks for testing!

closed 2017-08-07 by forkineye