forkineye/ESPixelStick #63: Multi Universe Multicast

mysecretstache, 2017-08-03:
I've been testing w/ an ETC Eos system, Luminair, as well as MAD Mapper, and they don't seem interface well with the Pixelstick Firmware.  I can't get the PixelStick to respond to any of these Pro Level aplications/hardware without putting the Pixel Stick into multicast mode which of course currently supports one universe.

None of these pieces of software have a unicast mode.  Is there planned support for multiple universes in muliticast?  I'm trying to control a 256 Pixel Panel with a single controller. using either an EOS lighing console or MAD Mapper.

forkineye, 2017-08-03:
Multicast for multiple universes was implemented a while back and works in my testing, however I just been testing with Vixen 3 and xLights as I don't have access to professional software or hardware.  What version of the ESPixelStick firmware are you using?

mysecretstache, 2017-08-03:
I'm using 3.0 Rc1.

Using any of my regular tools, I can't get any output on the PixelStick unless I select Multicast, but I still only see response on Universe 1.  Nothing is carrying over into Universe 2.  I have confirmed w/ sACNView that I am sending data, and I see data on my DMX Gateways.

mysecretstache, 2017-08-03:
I am running this on an Adafruit Feather ESP Breakout board since I have a few dozen of them laying around from a canceled project.  But I don't think that would make a difference.

I also confirmed that I can control all the pixels through the test modes in the web interface.

forkineye, 2017-08-03:
What are you using for an AP and are there any settings for IGMP within it?  Anyway you can verify the IGMP subscriptions on the AP or open up multicast / disable IGMP snooping for testing?

mysecretstache, 2017-08-03:
I'm using an airport express so I don't have a ton of control over
settings? But it is waht I've used in show conditions in the past.

I should have a chance to test w/ a ubiquity setup in the coming month.

forkineye, 2017-08-03:
I'll do some testing and IGMP sniffing tonight to see if anything has broken recently.  Sounds like the AP is ignoring the IGMP subscriptions for some reason.  Just to clarify, you have other wireless devices associated to the airport express that are receiving multiple universes per device via multicast properly?

mysecretstache, 2017-08-03:
Good point.

In the past I've connected Ethernet devices to the AP but done a wireless
uplink from the controller for a couple of music tours where we didn't have
a good hardline connection.

My only project involving 100% wifi. each receiver was an isolated
universe.

I'll try with another AP as soon as it isn't out on a show.

forkineye, 2017-08-03:
Note that if your console / e1.31 source is wireless as well, you'll have all sorts of lag issues and dropped packets going beyond a single universe (if on the same AP) due to the simplex nature of 802.11.  You need to have a wired source when using ESPixelStick devices for optimal performance.

mysecretstache, 2017-08-03:
That I expected. From past experience.

forkineye, 2017-08-03:
Found a bug in that was causing the IGMP subscriptions to be all jacked to due moving some code around.  Didn't notice it on my end as I don't use IGMP snooping.  Just pushed a fix - https://github.com/forkineye/ESPixelStick/commit/3bcbcfa13d79f4e101edf25cd0809518e51691ff.  Let me know if it solves your issue.  Thanks!

mysecretstache, 2017-08-03:
> snooping. Just pushed a fix - 3bcbcfa

Will do. I probably won't have a chance to test till next week.

forkineye, 2017-08-05:
Found another bug while testing in lwip withing the ESP8266 Arduino project.

mysecretstache, 2017-08-09:
Works Now  Thanks for your help.  If you ever want to borrow a license of Mad Mapper for testing, let me know

forkineye, 2017-08-09:
Great to hear, thanks! I may take you up on that in the future.

closed 2017-08-09 by forkineye