forkineye/ESPixelStick #64: Current config preserved after OTA reflash

penfold42, 2017-08-06:
It seems that an espixelstick with a current valid config will retain this after an over the air reflash.

Is this a feature ?

Could we make it optional ?

forkineye, 2017-08-07:
It's a feature.  After everything is flashed, ```config.json``` is written back out with the current configuration.  If something is missing from the config file after reboot due to new features, its needs to be initialized from the sketch and ```validateConfig()``` is a good place to do it.  What's the use case for overwriting the config?  It would still need to retain the WiFi section so connectivity isn't lost.

penfold42, 2017-08-07:
I'm ok with this behaviour - it just caught me out when I made some changes to config.json while reflashing OTA to a device with no USB serial.

closed 2017-08-08 by forkineye