forkineye/ESPixelStick #65: Need to verify if websocket data is revceived

forkineye, 2017-08-18:
Sometimes, configuration data fails to show up in the web UI.  Root cause is unknown, however configuration receipt verification / re-querying should be done to prevent this from happening.

cinoan, 2017-08-18:
Traced it to ESPAsyncWebServer before but didn't get a fix and haven't had chance to look recently - https://github.com/me-no-dev/ESPAsyncWebServer/issues/138

Sending one message, e.g
`ws.send('E1#G1#G2');`
and handling that rather than the current
`ws.send('E1'); // Get html elements
ws.send('G1'); // Get Config
ws.send('G2'); // Get Net Status`
would probably avoid the issue.

forkineye, 2017-08-18:
@cinoan Appreciate the help, thanks!  I'll look into it.

forkineye, 2017-08-30:
@cinoan I'm having issues re-creating this locally.  It looks as if it may have fixed here - https://github.com/me-no-dev/ESPAsyncWebServer/pull/144.  They have since fixed the IE issue as well so my fork of ESPAsyncWebServer should no longer be required.  Any chance you can see if the latest ESPAsyncWebServer upstream still has the issue?

cinoan, 2017-09-07:
@forkineye Sorry I hadn't got back to this before. Just tried updating the ESPAsync libraries and still getting the same issue with messages being missed. Short on time at the moment but will try to find some to look at it.

PR #78 (2017-11-10): Add Websocket Queue
Queue websocket requests from the client to avoid an issue with the server not receiving/responding to all requests if sent together.  Fixes #65
closed 2017-11-10 by forkineye