forkineye/ESPixelStick #68: Gamma curve doesn't seem correct

Jchuchla, 2017-09-06:
While diagnosing another issue, I was using a value of 25,21,2 (set by MQTT) The light did not light at all. if I disabled gamma correction, the light worked properly. So I knew that the gamma adjustment was adjusting these values down to zero.  25 is quite far from zero to get no action on the light.  I would be more understanding if 2 resulted in no light.  A proper gamma curve should light the light with all non-zero values.  It appears that a LUT is being used similar to that published by adafruit: https://learn.adafruit.com/led-tricks-gamma-correction/the-quick-fix  This really isn't a good gamma curve for lighting because there's too much deadband.  This method is more appropriate for graphics.  A more in depth practical guide on gamma adjustment can be found here: https://hackaday.com/2016/08/23/rgb-leds-how-to-master-gamma-and-hue-for-perfect-brightness/

Jchuchla, 2017-09-06:
A side note.  Because of the above, I would think that gamma correction should be off by default.  since the ESPixelStick is a great beginner product, it's best to avoid confusion with beginners related to why so many values result in no light.  gamma correction is more appropriate for video type displays, which are not a beginner project, and not necessarily the prime application for the ESPixelStick.

penfold42, 2017-09-06:
I don't know how fast float operations are on the esp8266 or how much bigger the sketch will be if we use things like pow()z

Maybe we should have a couple of gamma tables and turn the toggle into a dropdown defaulting to gamma 1 (off)

Jchuchla, 2017-09-06:
Perhaps you can calculate the LUT when the gamma value is set. This would be a compromise between doing it with float math on the fly and using a static table (or multiple tables). Who cares if it takes several seconds to calculate the new gamma table when you first select a new value.

--Jon Chuchla--

forkineye, 2017-09-06:
You're correct in that gamma in its current state is just a LUT with some dead space at the front like the link from Adafruit.  I had originally intended calculating the table based off a user gamma value, however pow() was broken back then.  I'll look into it once 3.0 is out.

penfold42, 2017-10-10:
Done
#73

penfold42, 2017-11-18:
Incorporated in #86 along with pwm support

closed 2018-07-23 by forkineye