forkineye/ESPixelStick #69: Add the possibility to configure several types of led according to their position

thisim, 2017-09-14:
Would it be possible to add the possibility in your software to be able to configure several types of led according to their position. For example 1 to 20 RGB and then 21 to 50 BGR then 51 to 70 RGB.

Thank's

penfold42, 2017-09-14:
It's possible but would likely be ugly to configure.

Can you remap the colour channels in your lighting source?

thisim, 2017-09-14:
I use xlights to drive the espixelstick. 
In xlights, I can choose the type of led ribbon, but not on a single object. 
I tried to do this in xlights, but this causes a color shift. 
I suspect it may take a long time to set up on espixelstick, but this is the only solution for me.
 If you could make me a version of your software with the ability to set each pixel type for each channel in each universe, that would be great.
 I would have liked to modify the program myself, but I'm nil in programming. 
Even if I need to add a config file manually, that suits me perfectly. 
Sorry for my English, but I'm French. 
Thank you.

bquesse, 2017-09-14:
You can absolutely achieve this with xLights.

thisim, 2017-09-14:
In fact in xlights, we can choose the type of led on a set. 
For example, if I have a window that is surrounded with a single garland composed of LED RGB and BGR, in xlights I can not choose the 2 types because the window is considered as a single garland. 
That's why I wanted to be able to choose the type of LED in the espixelstick.

 I also have another problem. 
In fact, I use 3 universes on the espixelstick. I configured each universe on 510 channels.
 I also configured 510 channels per universe in xlights. 
It is from the first LED of the second universe that there is a pixel shift. 
The first channel of the second universe lights up the first pixel in red and the second in green + blue if I ask the first pixel of the second universe in white.
Then, it shifted me all the other pixels. 
Maybe it's a bad setting of xlights or espixelstick, but I do not see where.

forkineye, 2017-09-15:
I won't be adding this functionality to the ESPixelStick code. Within xLights, I recommend making separate models for the elements so that you can assign RGB order to them within xLights.  As for the boundary issue you're having, are you on 3.0-rc3 and have you set Universe Boundary to 510 on the ESPixelStick? It's 512 by default.

thisim, 2017-09-17:
In xlights, that's what I did. 
I separated my elements in two when they were composed of 2 type of garlands. 
Otherwise, I was using version v2 as it was for Christmas 2016. 
I have already updated my espixelstick with the latest version. 
I will soon reinstall the LED garlands on my house to try again.

thisim, 2017-09-17:
I configured the espixelstick and xlight on 510.
 I thought it was not possible to have more than 510 per universe which makes 4 universes of 170 led.

madsci1016, 2017-09-17:
Data is data. 512 channels per universe works fine with no noticeable effects. My show runs like that for 2 years now.

thisim, 2017-09-17:
Ok, thanks for the quick response.
 I'll try again. I probably made a bad adjustment. Now with the test mode integrated in the new firmware I will be able to test more easily.
 I thought it was 510 because as there are 3 channels per LED, 512 is not divisible by 3. 
I will try again quickly.

thisim, 2017-09-17:
I made a video at Christmas 2016. 
Here is the link: https://youtu.be/APp21fl6uL4
 if you look at the bottom of the middle window, you will see that there is a pixel that does not work well. 
There are others, but it is not visible on the video. This problem begins at the first channel of the second universe.

madsci1016, 2017-09-17:
512 will work ok, so will 510. But it will only work if both sides (the transmitter and receiver) agree to use the same convention. Data from all universes is dumped into the same buffer, so it won't matter practically if the data for one pixel came from two different universes.

thisim, 2017-11-15:
With the new firmware RC3, everything is OK, I no longer have any pixel offset. Thank you fornikeye for this great program.

closed 2017-11-15 by thisim