forkineye/ESPixelStick #70: APA102 support

thommy86, 2017-09-19:
The Github version of the feature-ws2801 wasn't able to compile, so I merged it with the master. This is working as expected now. But as I am using APA102 I was wondering if you are going to add support for this. The leds are full white now and I am not able to control them.

I am using a Wemos d1 mini. Connected data to gpio 2 and the clock pin to TX.

GeorgeIoak, 2018-12-25:
I found this project that is based on ESPixelStick but extends it to include APA102 (SPI) based LEDs. I haven't tried it yet but it should be possible to extend ESPixelStick with Thomas's code to add support for the other LEDs.

[APA102 LED Stripe controlled by ESP8266 using E1.31 / SACN](https://hoeser-medien.de/2017/12/apa102-led-stripe-controlled-by-esp8266-using-e1-31/)

forkineye, 2020-12-01:
With only one SPI accessible to us on the ESP8266 platform, we chose to use it for SD card usage instead of clocked pixel support, so this won't happen.

closed 2020-12-01 by forkineye