forkineye/ESPixelStick #71: Feature Request:  Relay or GPIO output

Jchuchla, 2017-09-30:
It would be nice to have a mode where the ESPixelStick code could be set up to be able to provide a digital trigger to be used for (external) relay control.  I'm thinking simple on/off control of a power circuit.  This would likely be more relevant in MQTT mode, but I suppose it could be equally valuable with an sACN data source. I realize that with the current ESPixelStick V2 boards, this is limited to only one output.  Maybe two, I don't know if the C line is connected to an output of the ESP.
If going down this route, a logical extension of the idea would be to also allow these to also be used as inputs.  In this case, sACN wouldn't be relevant, but it could certainly publish an MQTT topic with their state.

penfold42, 2017-10-01:
Could you abuse my pwm branch/pull request ?

Jchuchla, 2017-10-01:
Lol. I’m sure I could abuse your branch. 
Do you have notes on what you included and how it works?

penfold42, 2017-10-01:
Gpio pins can be mapped to DMX channels.

If you have a mosfet driver or a simple led it's brightness will be set based on the DMX value.
0 is off, 255 should be fully on.

A relay would likely buzz at in between values.

You'll need to enable it in the source code (like selecting ws2811 or serial options).

It's configured via the web UI and there are options to:
- globally enable/disable it
- set the pwm frequency
- map which dmx channel is which gpio
- invert the output (255 is off)

Jchuchla, 2017-10-01:
Does it work with the MQTT configuration?  It seems like the MQTT implementation is just a test pattern that fills the dmx buffer with the 3 values over and over. So I'd imagine it would.
Is this branch in sync with the current master?
How many GPIOs does it support on the current ESPixelStick 2.0 hardware?

penfold42, 2017-10-01:
It should work with Mqtt.

It should be in sync - there's a pull request awaiting merge.

It can support any esp8266 gpio - from memory I blacklisted gpio 2 cos it will likely clash with 99.99% of users

penfold42, 2017-10-11:
I can confirm it DOES work with mqtt

penfold42, 2017-11-15:
When #66 is merged this can be closed?

Jchuchla, 2017-11-15:
66 mostly covers this request. However It would be nice to also have an option to lock the output to full on and full off So that you can’t accidentally damage a connected device that can’t handle a dimmed PWM output.

--Jon Chuchla--

penfold42, 2017-11-15:
A checkbox to make each of them digital ?
And use 128 as a threshold value ?

penfold42, 2017-11-16:
Done (in theory!)
4b9e57fa6ef70976a216210066294cd3a71a8797

penfold42, 2018-07-12:
And now removed from this repo as of today.

I’ll be maintaining this support in my repo along with other features:
https://github.com/penfold42/ESPixelBoard/tree/ESPixelBoard?files=1

forkineye, 2018-07-12:
@penfold42 I'll make a note of it in the readme and changelog with a link to your fork. Thanks!

closed 2018-07-12 by forkineye

priyankchouksey, 2018-11-27:
> And now removed from this repo as of today.
> 
> I’ll be maintaining this support in my repo along with other features:
> https://github.com/penfold42/ESPixelBoard/tree/ESPixelBoard?files=1

Hi,
I am looking to add relays on few GPIO, Can you please let me know what to set in the DMX/GPIO UI and what data from sequence it would be expecting.

Thanks in advance for help