forkineye/ESPixelStick #72: Struggling to compile

nrudolph7, 2017-10-04:
Hello!

I've previously used V1.3 of your firmward on some D1 Mini's (esp8266-based) and was able to compile in the arduino ide with no issue.

I have installed all of the libraries that are listed on the github page, but keep running into errors when compiling in the ide. Specifically, it is looking for an E131async.h file to include but can't find one. Any suggestions? Would this be located in a library and somehow the library didn't install properly?

Is it also necessary to follow up build/flashing with uploading web files to the device, or if I'm custom compiling in ide and uploading that way, are they already included? When I compiled and used V1.3, everything was done with the compile and upload button in the ide and it just worked from the get go.

Also, am I able to flash the pre-compiled firmware for a PixelStick on one of these D1 Mini's? Or do I need to custom compile my own build?

Thank you for your help!

nrudolph7, 2017-10-04:
Update: I believe there is an error in the .ino file.

In the .ino, Line 46 is: #include <E131Async.h>

I changed it to: #include <ESPAsyncE131.h>

This removed the error I was receiving.

However, I'm now receiving this error:

C:\Users\User\AppData\Local\Temp\arduino_build_592289\sketch\EFUpdate.cpp:23:18: fatal error: util.h: No such file or directory

 #include "util.h"

                  ^

compilation terminated.

Any suggestions?

forkineye, 2017-10-04:
Looks like you need to update your source tree. It's defined here - https://github.com/forkineye/ESPixelStick/blob/master/ESPixelStick.ino#L41

closed 2017-10-19 by forkineye