forkineye/ESPixelStick #75: trying to compile

fastfourier666, 2017-10-24:
Hey there,

I'm having some trouble compiling ESPixelStick-3.0rc3 using Arduino 1.8.5 and ESP8266 2.4.0-rc2. I installed all the required libraries straight from github. After making the change to `#include <ESPAsyncE131.h>` I'm getting a lot of errors related to `RingBuf *pbuff`

I couldn't get them to look reasonable here so [here is a pastebin of the whole thing](https://pastebin.com/nFKeEYPV)

forkineye, 2017-10-25:
What are you having to change?

penfold42, 2017-10-25:
Multiple libraries were found for "ESP8266mDNS.h"

Do you multiple copies of other libraries ?

forkineye, 2017-10-25:
@fastfourier666 I just looked at the source archive for 3.0rc3 and there's an issue with it. For some reason, the sketch wasn't tagged properly and is an older version. Recommend pulling from the latest master.

fastfourier666, 2017-10-25:
@forkineye I was changing the #include from "E131Async.h" to "ESPAsyncE131.h" to match the example in the ESPAsyncE131 library.

I would like to try out the new features in v3. Should I be looking at using an earlier commit of the ESPAsyncE131 library, or [this library](https://github.com/forkineye/E131)?

The v2.0 release of ESPixelStick works great though, thanks!

fastfourier666, 2017-10-25:
@penfold42 Only that one I think. I've since got rid of it.

forkineye, 2017-10-25:
@fastfourier666 I had edited my previous reply. You need to pull the latest from the git master as the libraries have changed since 3.0-rc3 was released.

fastfourier666, 2017-10-26:
Sorry, i'm confused (and a bit new to this!) You mean I should pull the latest ESPAsyncE131?

forkineye, 2017-11-01:
@fastfourier666 No, that latest ESPixelStick code. Compile against the latest pull, not the 3.0-rc3 release.

closed 2017-11-17 by forkineye