forkineye/ESPixelStick #77: Max Espixelstick on wifi(with specific ip & mac)

Kamal-Sonani, 2017-11-02:
◆ Wifi router's defult gateway (192.168.1.1)
◆ 50 espixelstick need to connect

**1:- Is it possible to make (1st of 50 )espixelstick set on fixed ip address x.x.1.2 ,x.x.1.3 and so on....** 

Ex:- (300mbps)Router's can register only 34 client's mac address for fixed ip address.
Can i set each espixelstick to DHCP fixed ip address without setting in router?
 If so, every time each espixelstick find his previews ip👍 .

**2:- disable to create AP, only try to join ssid** 

Problem ⤵
- Each time when i start whole show, During that time espixelstick find ssid to join network but router takes 40 to 50 second to wake up and working.So esp goes in AP fall back mode.

penfold42, 2017-11-02:
1. You can just set static addresses on each espixelstick - don’t use dhcp

2. You can disable AP fallback but when the timeout is exceeded it will reboot and this might not help you.
It sounds like your AP is either too distant or just a terrible AP. 40s is ludicrous

forkineye, 2017-11-13:
I would recommend just disabling AP fallback mode.

closed 2018-07-23 by forkineye