forkineye/ESPixelStick #79: Bookmarks to specific web pages don’t work

penfold42, 2017-11-12:
If I launch http://10.0.0.68/#test directly, it ignores the #test and drops me in the home page

cinoan, 2017-11-12:
Adding
`    //if page location has a hash then click the corresponding nav link to change to that page.
    if (location.hash !== '') $('ul.navbar-nav li a[href="' + location.hash + '"]').click();`
into script.js before `// DHCP field toggles` works around this issue.

I haven't done it as a pull request as I'm not sure it is the best way to do it, or if it should be done. The sections are just tabs rather than pages and so I'm not sure if it would be better if the hash was removed and users were just always directed to the home page on load. Trying to make the tabs act like pages is probably never going to be perfect and so a bit jarring.

penfold42, 2018-11-03:
Thanks - I’d forgotten about this issue I openend !

Done as part of #136

closed 2018-11-03 by penfold42