forkineye/ESPixelStick #82: Merge websocket heap, RSSI and uptime query

penfold42, 2017-11-13:
Any downside to this?

Small performance gain and frees up some letters!

cinoan, 2017-11-13:
I think the only downside would be for a different client that was only interested in one stat, say a monitoring system that is just checking the uptime or a status page for multiple ESPixelSticks where you might split the different stats over different tabs. But I don't think those are too likely at the moment.

If you were to merge them (& e131 stats could be included) then I think it would be best to combine them into one websocket message back rather than just changing it so one request triggers all 4 current responses. That would help avoid any other websocket issues and reduce the network traffic.

penfold42, 2017-11-13:
I was going to write a new call that incorporates them all in one csv response and then update the webUI to use that instead.

Json would be nicer but I’m not convinced the flash, ram and code overhead is worth it

closed 2017-11-17 by penfold42