forkineye/ESPixelStick #84: Change GPIO output

thisim, 2017-11-13:
I bought several modules of this type and the GPI02 is not used.

https://m.fr.aliexpress.com/s/item/32765066001.html?trace=storeDetail2msiteDetail

I wanted to use the GPIO3 output to control the pixels.
I tried to edit the file espixelstick.h but I was not successful.
When I check the sketch , I have plenty of error messages. 
I do not know the programming arduino well. Would it be possible to send me a link to a firmware with the modified GPIO. 
I want to use GPIO3 instead of GPI02.
A firmware with automatic programming like this one (ESPixelStick v3.0-rc3 Firmware)   would be perfect.
If anyone could help me, that would be very nice.

thisim, 2017-11-15:
after several tries, i finally managed to reprogram the esp01 with the file espixelstick.h with the data pin GPIO3 instead of gpio2. The programming goes well, but the gpio does not change. LED control is always done by the GPI02. In addition, when I want to access the espixelstick configuration web interface, this marks me page not found. Is there another file to modify to change the Gpio? I just replaced the value data pin at 3 instead of 2 at the beginning of the file espixelstick.h

forkineye, 2017-11-15:
You can't use a different gpio pin for ws2811 pixel output as the pixel stream is generated with the uart. Only gpio2 is supported.

closed 2017-11-15 by forkineye

thisim, 2017-11-15:
Ok, thanks a lot for the answer.

jdanbru, 2019-10-23:
Is there a specific pin we HAVE to use on the Huzzah (ESP-12S?)

Is it also GPIO2?