forkineye/ESPixelStick #87: RECORD E1.31 STREAMING

Kamal-Sonani, 2017-11-16:
Is it possiable to record e1.31 signal stream ?

For use offline instead of with xlights or vixen or any other software.

& If it possiable, i can control only 700 led ?  Or merg two or more espixelstick with some communication and extend pixel driving size?

penfold42, 2017-11-16:
No, not at the moment.

there’s probably not enough storage space to record a stream in the current hardware.
There a 1 MiB flash chip, 128k is used for the web server and 400k is used by the code (at the moment) so there’s about 400k left but this is needed for over the air updates.

500 lights needs 1500 bytes and at say 25 frames per second you’re at 37k/sec so you only get about 10 seconds worth

Kamal-Sonani, 2017-11-16:
what about sd card?

penfold42, 2017-11-16:
Thinking some more, we could use that OTA space but you’d have to preserve it yourself if an OTA update was done

penfold42, 2017-11-16:
Sdcard could work but only if the write speed was fast enough. There’s only about 20k ram free at the moment and you’d need this to buffer the data

Kamal-Sonani, 2017-11-16:
if matter is about speed and space ,
Esp32 should enough.

what you think?
all n all i want to do this at any cost😂😂😂

penfold42, 2017-11-16:
I think it’s simpler to pre load the light show a file than record it from the network

aususer, 2017-11-17:
fty: you might want to check out: http://docsteer.github.io/sacnview/
there are other apps out there that can do exactly this.. e1.31 is multicast - so you don't have to have a specific device to do this - anything on the subnet could potentially do this. 
eg. I believe someone else out there is trying to do a similar thing on raspberrypi... 
hope this helps

trigx300, 2017-11-21:
I'm interested in working on this. I'm not a total noob, but I'm not familiar with some of this. I don't have the ESPixelStick hardware I'm just programming dev boards and the esp8266 with minimal resistor hardware and a step-down buck converter. I'm thinking of using the D1 mini as a starting point for this and adding a micro sd shield. The main issue seems to be buffer memory and sd R/W speed. My plan is to start with a sketch that can simply WPS and record E1.31 packets to the SD as well as play them back. I'd be removing most of the code from this project in order to free up memory. My assumption is future hardware (ESP32) will allow us to merge the two easily.

penfold42, 2017-11-21:
I’d focus on playback - leave recording to something else

Kamal-Sonani, 2017-11-28:
one step ahead 
its may very rude for you😂

i just record espixelstick's D4 pin streaming signal (output) ,as audio in voice recorder and save as .wav .mp3

its works great😎 
but some time flickering appear cause of noice

not perfect solution but its works for me now .
Any Suggestion?

Kamal-Sonani, 2017-11-28:
@penfold42 but how?

Kamal-Sonani, 2018-02-14:
https://youtu.be/Zn1N7YUXMdQ. 

According this video sd card playback is done with record artnet data

Any other idea for direct play? Scriptwise

forkineye, 2018-07-23:
Closing this one as it'd never happen on the minimal platform for this project (ESP-01).

closed 2018-07-23 by forkineye