forkineye/ESPixelStick #88: NODEMCU 1.0 upload

aususer, 2017-11-17:
HI,

I can't buy one of the hardware devices (as Im in AUST) but I have a stack (10+) of nodemcu 1.0 (arduino with ESP8266 on them) so I want to install this code on one of them to do this (yes.. acknowledge its overkill)... 

I've managed to upload the ide (took a while to find out I needed 2.4.0-rc2 but got there) and I can now see the IPaddress in serial monitor; can get the "page not found" if I http to it; and more response if i http to /config, so I know the arduino/esp part is working.. and my pc can see it..
`ESPixelStick v3.0-rc3
- Configuration loaded.

Connecting to BigPond3944 as null
Connecting with Static IP..........
Connected with IP: 192.168.0.32
- Web Server started on port 80
- Unicast port: 5568
- Listening for 1 channels, from Universe 1 to 1
`

I am now at the point where I have to do the "gulp" upload via windows 10.... and I've hit a wall.
I can't for the life of me work out how to (on windows) do this "gulp" and upload into the hardware.
Ive installed the upload code and it is working in my arduino 1.8.5, but I am completely lost at the
"compress with gulp and upload" bit
has someone done this on a non-forkineye hardware setup before?

FWIW: my WS2812's will work with the arduino - pixel testing code proves that so its just an installation problem I have now to get this final step.
Thanks in advance.

penfold42, 2017-11-17:
Gulp is just the preprocessor to optimise the web server content.

This tool https://github.com/esp8266/arduino-esp8266fs-plugin is used to do the upload

penfold42, 2017-11-17:
Sorry, didn’t read tour post clearly.

Seen this ?
https://github.com/forkineye/ESPixelStick/blob/master/html/README.md

Kamal-Sonani, 2017-11-17:
try with direct bin file instead of installing from arduino

aususer, 2017-11-17:
You are all awesome... I've got it to work.
FWIW:  I am not sure what Kamal-Sonani means by "direct bin file" but now I better understand the "node.js" bit... I feel daft (in hindsight) for asking!
None the less its now working thanks to your help.
Can I suggest an edit/addition in the GULP readme file for more clarity and more specific on the installation process?

<readme.md>
++
**GULP and Node.JS in windows 10**
You need gulp (running under node.js software) to package all of the data/html and css files into a format that can be uploaded and used by the the ESP8266. If you don't do this step you will only ever  see "page not found" messages in your browser.

To do this under windows requires the installation of  "NODE.JS" software, download from https://nodejs.org/en/. You don't need the latest version - just the most stable.

Once installed, open your Windows Start Menu and find a new app called "Node.js command prompt". This will open a black screen that looks similar to a DOS/CMD prompt. 
Ensure you know where your espixelstick code is installed, as you will need to change directory into the folder for the gulp to work
(in the below example, the GIT was downloaded as a zip and extracted into **c:\download\espixelstick** it may be different for your installation)
Your window should look similar to this:
            _Your environment has been set up for using Node.js 8.9.1 (x64) and npm.
            C:\Users\[user]>_**cd c:\users\[user]\downloads\espixelstick**

Now you need to install the "gulp" modules into node.js to do all of the work: 
            _C:\Users\[user]\Downloads\ESPixelStick>**npm install -g npm**
            C:\Users\[user]\AppData\Roaming\npm\node_modules\npm\bin\npx-cli.js
                 + npm@5.5.1
            added 460 packages in 125.446s_

Now to install the global components for gulp client into node.js:
            _C:\Users\[user]\Downloads\ESPixelStick>_**npm install -g gulp-cli**
             _C:\Users\[user]\AppData\Roaming\npm\gulp -> C:\Users\
              [user]\AppData\Roaming\npm\node_modules\gulp-cli\bin\gulp.js
              + gulp-cli@1.4.0
              added 139 packages in 37.122s_

Install all the modules (you may get some warning errors that you can ignore):
              _C:\Users\[user]\Downloads\ESPixelStick>_**npm install**
             _npm WARN deprecated gulp-clean-css@2.4.0: breaking changes from clean-css 4.x. Please install 
             gulp-clean-css 3.x
             npm WARN deprecated minimatch@2.0.10: Please update to minimatch 3.0.2 or higher to avoid 
             a RegExp DoS issue
             npm WARN deprecated minimatch@0.2.14: Please update to minimatch 3.0.2 or higher to avoid 
             a RegExp DoS issue
             npm WARN deprecated graceful-fs@1.2.3: graceful-fs v3.0.0 and before will fail on node releases 
              >= v7.0. Please update to graceful-fs@^4.0.0 as soon as possible. Use 'npm ls graceful-fs' to 
              find it in the tree.
              npm notice created a lockfile as package-lock.json. You should commit this file.
              added 377 packages in 64.54s_

Run GULP to process the files and make the data/www folder - (make sure it is still in the base installation folder of your espicelstick as that is where the gulpfile.fs script file is located.)
                    _C:\Users\[user]\Downloads\ESPixelStick>_**gulp**
                    _[21:44:04] Using gulpfile ~\Downloads\ESPixelStick\gulpfile.js
                    [21:44:04] Starting 'clean'...
                    [21:44:04] Starting 'html'...
                    [21:44:04] Starting 'css'...
                    [21:44:04] Starting 'js'...
                    [21:44:04] Starting 'image'...
                    [21:44:04] Finished 'clean' after 141 ms
                    [21:44:04] Finished 'image' after 71 ms
                    [21:44:05] Finished 'html' after 1.4 s
                    [21:44:05] Finished 'css' after 1.34 s
                    [21:44:10] Finished 'js' after 6.22 s
                    [21:44:10] Starting 'default'...
                    [21:44:10] Finished 'default' after 102 μs_

Now you are done!

Close Node.JS application and open your arduino editor. Ensure "serial monitor" is **not** running ( close it if you do) and go to "tools|ESP8266 Sketch Data Upload", and let it run (it may take some time to complete - the percentage will increase during the process). 
When it reads 100% done, reboot your esp8266 (hit the reset button), and open your serial monitor to check network setup is working (if you statically defined your SSID/passcode in the .ino)

open your browser and link to that IP address to be presented to the html configuration pages.
Uploading is now complete

++
</readme.md>

closed 2017-11-17 by aususer