forkineye/ESPixelStick #89: Web save issues

Mat-Moo, 2017-11-17:
Ok, I'm quite new to this, only my second season, but have been a programmer for 40 odd years. I'm using the ESPixelStick Flash tool to program my modules, an all works like a dream. It then hooks onto my wi-fi and I can connect and run tests on pixel strings and all works great... till I change any setting and hit save, then the module seems to give up. If I reboot I just get a steady blue flashing light and can no longer see it till I reprogram it. I've tried latest 3rc and also fallen back to 2rc same problems :( any clues? (This is with 2 different esp wifi boards, although same brand)

forkineye, 2017-11-17:
What settings are you specifically changing. Also, what are you seeing in the flashtool serial output when this happens?

Mat-Moo, 2017-11-17:
Upload
`-= Building SPIFFS Image =-
 bin/win/mkspiffs.exe -c spiffs/ -p 256 -b 4096 -s 131072 firmware/spiffs.bin
/config.json
/www/esps.css.gz
/www/esps.js.gz
/www/index.html.gz

-= Programming ESP8266 =-
 bin/win/esptool.exe -cd none -cb 115200 -cp COM4 -ca 0x000000 -cf firmware/pixel-3.0rc3-1m_128k.bin -ca 0xDB000 -cf firmware/spiffs.bin
Uploading 357824 bytes from firmware/pixel-3.0rc3-1m_128k.bin to flash at 0x00000000
................................................................................ [ 22% ]
................................................................................ [ 45% ]
................................................................................ [ 68% ]
................................................................................ [ 91% ]
..............................                                                   [ 100% ]
Uploading 131072 bytes from firmware/spiffs.bin to flash at 0x000DB000
................................................................................ [ 62% ]
................................................                                 [ 100% ]

-= Programming Complete =-
`
After upload

`Connecting with DHCP.......
Connected with IP: 192.168.2.101
- Web Server started on port 80
- Listening for 510 channels, from Universe 1 to 1
- Unicast port: 5568
* WS Connect - 1
- Listening for 510 channels, from Universe 1 to 1
* Con`

So that is the log from programming it up, I can connect with vixen and use it and the embeded tests...

Then I changed the pixel output from RGB to GRB and hit save... all ok till reboot. That's all that happens and then the chip just flashes a blue led :(

In boot mode it also spat out
`
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

loa`

Admin screen shows

> Administration
FW Version
3.0-rc3 (20170805)
Used Flash Size
1048576
Real Flash Size
1048576
Flash Chip ID
146085

Thanks for your support, much appreciated in my battle against time!!

forkineye, 2017-11-17:
Did you reboot it from the UI or are you power cycling it?

Mat-Moo, 2017-11-17:
Think I’ve tried both, but I would say mostly power cycle

penfold42, 2017-11-18:
You MUST power cycle or hit a reset button after any serial upload.

If you don’t, a software initiated reset usually hangs it

Mat-Moo, 2017-11-18:
Ok, programmed the ESP, then power cycled. Connected and changed to static IP, same issue:-
`Connecting with DHCP...........
Connected with IP: 192.168.2.101
- Web Server started on port 80
- Listening for 510 channels, from Universe 1 to 1
- Unicast port: 5568
�... Load of junk
ESPixelStick v3.0-rc3 (20170805)
- Configuration loaded.

Connecting to chrismoo as esps-41f933
Connecting with DHCP........
Connected with IP: 192.168.2.101
- Web Server started on port 80
- Listening for 510 channels, from Universe 1 to 1
- Unicast port: 5568
* WS Connect - 1
- Listening for 510 channels, from Universe 1 to 1
* Configuration saved.
* WS Disconnect - 1
*** WiFi Disconnected ***

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
�
ESPixelStick v3.0-rc3 (20170805)
- No configuration file found.
- Listening for 1 channels, from Universe 1 to 1
* Con`

penfold42, 2017-11-18:
How are you powering it ?

The 3,3v from a typical ftdi USB serial adaptor is insufficient

Mat-Moo, 2017-11-18:
12v supply to my board and using an onboard regulator (uk board based on this), not via usb.

penfold42, 2017-11-18:
Rst cause of 2 is the reset pin from memory.

What modules do you have and do you have the right pull up resistors setup on rst, ch_pd, and the gpio lines ?

Mat-Moo, 2017-11-18:
At a loss, can only think its an esp module issue. If I take one of my old module ESPs (v1 firmware) and install it into a new board, I can change a setting, save and all good when I reboot. But everytime I save on either of the two new boards I have bang, loses config :( so I don't think its a resistor,board issue, but a esp issue.
https://www.ebay.co.uk/itm/ESP8266-Serial-WIFI-Wireless-Transceiver-Module-Send-Receive-AP-STA-Arduino-LQ/222694237921?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649
thinking of getting https://www.amazon.co.uk/gp/product/B0728CBGKT/ref=ox_sc_act_title_1?smid=A1HPO255OC5X3C&psc=1 to test

penfold42, 2017-11-18:
Try erasing the flash on the 2 boards that don’t work.

What size flash chips do they have ?
The admin page will tell you what it thinks the chip is (and what the code was compiled to expect)

forkineye, 2017-11-18:
I've heard this same thing happening to one other user and all I could trace it to was their module.  It was an ESP-01S knockoff with AI-Thinker misspelled. Most likely it was probably a junk flash chip.  SPIFFS does wear leveling which would explain why it may seem corrupted after saving config with a flaky flash.

Mat-Moo, 2017-11-18:
Is there a way to test? Should I get the ESP's I linked to on Amazon to try?

forkineye, 2017-11-18:
Can you post a close-up of the module where we can ID the flash chip on it?

Mat-Moo, 2017-11-18:
[image]

forkineye, 2017-11-18:
Oh, that rings a bell now. I was thinking it was an ESP-01S, but it was an ESP-01.  That's the exact same module the other user was having an issue with. Same "AI-Could1nside" silkscreen, fat traced antenna, and flash chip.  The real AI-Thinker ESP-01's look like this:
[image]

Mat-Moo, 2017-11-18:
Ok bought some more from Amazon will feedback tomorrow :(

sjernigan, 2017-11-18:
I'm having a similar problem.  Similar configuration as OP.  The test work after a flash until I try to power cycle, reset, or save.  Then it appears to lock up.  FWIW, the silkscreen on my ESP module says "AI-Cloud inside".

Mat-Moo, 2017-11-19:
Good news @forkineye new esp modules look much better and working like a dream! Thanks ever so much for your support!!
As a side note, maybe the java programmer should add a message after programming complete to power cycle the esp module? 

As a side note, I do quite a bit of JS and I think quite a few bytes can be saved by some code optomisation and library swaps, e.g.
http://zeptojs.com/
```
    $('#gateway').keyup(function () {
        wifiValidation();
    });
    $('#ip').keyup(function () {
        wifiValidation();
    });
    $('#netmask').keyup(function () {
        wifiValidation();
    });
```
to
`
    $('#gateway,#ip,#netmask').keyup(function () {
        wifiValidation();
    });
`
and
`if ($(this).is(':checked')) {
            //$('.mqtt').prop('disabled', false);
            $('.mqtt').removeClass('hidden');
       } else {
            //$('.mqtt').prop('disabled', true);
            $('.mqtt').addClass('hidden');
       }`
to
`
            $('.mqtt').toggleClass('hidden', ($(this).is(':checked'));
`

There is a few other things, Not got time right now to do a PR, but I now every byte counts, is this of any help/interest?

penfold42, 2017-11-19:
Glad you had better success with the newer modules.

I’m always interested in cleaner shorter code !

forkineye, 2017-11-20:
@Mat-Moo Bytes count and pull requests are always welcome. Thanks and glad you got it working!  @sjernigan Have you had a chance to try any other modules?

sjernigan, 2017-11-20:
Thank you for asking.  I have some arriving this week and will test as soon as I am back from the holiday.

sjernigan, 2017-11-28:
**SOLVED** 
The problem was that i was unplugging the FTDI cable from my computer but leaving it connected to the ESP module.  I don't know why that would cause it to hang on the reboot but the problem is easily resolved.  In other good news, both my "real" module (those with the "AI-Cloud inside" silkscreen) and my fake ones work just fine.  Thanks for the help.

penfold42, 2017-11-28:
My guess is unplugging it may have pulled gpio 0 low which would put it into upload mode on the next reboot

Neilw2017, 2017-12-07:
Having issue with nodemcu all works fine apart from can only seem to address a maximum of 170 leds am I missing something??

sjernigan, 2017-12-07:
170*3 = 510...512 is the limit for a DMX universe I think

Neilw2017, 2017-12-07:
But is it not possible to receive more than one universe worth of data via a node mcu???

forkineye, 2017-12-07:
@Neilw2017  Yes, just tell it how many pixels you have in the configuration front-end. It'll listen to consecutive universes.

Neilw2017, 2017-12-07:
Just stops at the 170th pixel on my matrix no matter what I try if I use the test it works perfect but vixen will only allow the first 170 this is so frustrating after spending a week trying to get it to work

Mat-Moo, 2017-12-07:
You need to configure vixen to use multiple universes, each universe has a max of 510 bytes. How many pixels in your matrix?

forkineye, 2017-12-07:
@Neilw2017 You sure you your Vixen outputs configured right and the matrix fully patched?

Mat-Moo, 2017-12-07:
[image]
Should look like that in Vixen

Neilw2017, 2017-12-07:
Matrix is 570 pixels laid out 19 * 30 have patched this with the even rows reversed element order too allow for the zig zag a total of 1710 patched outputs 4 universes showing 
1 start 1       size 510
2 start 511  size 510
3 start 1021 size 510
4 start 1531 size 510 
Multicast WiFi

Mat-Moo, 2017-12-07:
Never had much luck with Multicast, so can't help much more :(

Neilw2017, 2017-12-07:
Can i use a single nodemcu  under unicast will try anything lol granddaughter is over weekend for her lights and at the min only 1 3rd are working :(

forkineye, 2017-12-07:
@Neilw2017 Post a screenshot of your vixen configuration and your ESPixelStick home page w/ the stats.  Do you have any managed switches between your Vixen 3 computer and access point? What AP are you using?  You can try using unicast, could be an IGMP issue on your network.

Neilw2017, 2017-12-07:
I am going to dig out another access point I had problems with this one in past never thought it would be that and the nodemcu is the only device apart from the laptop which is connected

forkineye, 2017-12-07:
@Neilw2017 Make sure your source (laptop) is hard-wired. 802.11 is simplex and you'll get buffer lag if it's wireless on the same AP.

Neilw2017, 2017-12-07:
[image]

[image]

[image]
Will be hard wired to this ap and WiFi to my other router
[image]
Sorry for the bad pics but screen capture does not work on laptop

sjernigan, 2017-12-16:
This thread is kinda drifting but I'd like to go back to the original issue..the failures after save.  I thought I had it resolved but I received 10 more ESP modules from a couple of different vendors.  They all exhibit the same behavior.    They work fine till I try to change the name and hit save (which seems to take two clicks for some reason before I get the 'configuration saved' toast.  After that, the tests no longer work and the pixel count shows as 0.33333333333.  Sometimes other pulldowns like the pixel type are empty.  My setup is to program in a PixelStick and then move the modules to either a Pixel Pop or Pixel Pop tiny for testing.

penfold42, 2017-12-16:
A trashed config file will do that.
As will one that exceeds the maximum file size.

What do you get if you browse to http://<yourpixelstick>/config ?

If you’re building from source, there’s an option to serve the raw config.json too (but it’s conmented out by default)

sjernigan, 2018-01-01:
I think you could be correct about my save issues being at least partially due to the particular module.  I've borrowed one from a friend and could save fine with the same programming and pixelpop setup.  I've ordered 4 more from 4 more different vendors and will report the results.  If you have a known good vendor, would you mind posting links.  I report on my vendors when they are all in.

forkineye, 2018-07-23:
This is an upstream Arduino ESP8266 issue regarding PUYA flash chips used on many 3rd party modules and has to be fixed there. There's a patch in the following issue which may help you with patching your own environment - https://github.com/esp8266/Arduino/issues/4061.

closed 2018-07-23 by forkineye

ryanjjennings, 2018-09-25:
Do the latest Travis builds have this upstream patch included?

forkineye, 2018-09-25:
> Do the latest Travis builds have this upstream patch included?

As far as I can tell, the core hasn't been patched yet.  Travis builds off the latest git pull of the core so once they patch it, it'll be included in the next Travis build.