forkineye/ESPixelStick #90: lag when multiple multicast universes are configured.

Jchuchla, 2017-11-18:
I was troubleshooting my flakes earlier today.  See this thread.
https://diychristmas.org/vb1/showthread.php?8360-slight-ESPixelStick-lag
my first 4 flakes were all completely within one universe. The same universe.  The 5th started in universe 1 and continued into universe 2.  I am multicasting both universes.  Pixels are GECE

I was thrown at first while testing.  I set flake 5 to use the same start address as #4 and the lag was still there.  But if I rebooted it, it then cleared up completely.  If I set it back to the address where it spans universes, the lag came back right away.  And it stays thru a reboot as well.

Troubled by this, I proceeded to test my stars.  I have 6 of those and they contain a ton more pixels.  these are 2811.  I have 6 ESPixelsticks driving the stars.  Initially I configured them so that in case of problem, I could bypass any ESPixelStick and pass data thru from the one to the left.  So each one left to right has the ability to control less one star.  
160px   count 588
100px  count 428
100px count 328
50px count 278
50px count 228
128px  count 128
This was causing lots of lag and irregularities.  But when I set them all to only control the pixels that will normally be connected, then it's much better.  Some of these are still spanning universes, but there doesn't appear to be any lag here.  I fcan probably repatch these so that each pixelstick is contained within a universe, but it seems to be working ok at the moment with the reduced count.

Furthermore, it seems that when they get laggy, they actually play out all of their data to catch up.  They shouldn't play data that's old in order to catch up, they should catch up with the most current packet and discard anything that it couldn't display.  That leads me to believe that it's doing fine in e1.31 reception, because the buffer did fill with data, but something's causing it to play out slower when it's spanning universes.

Techdude06, 2017-11-22:
What wireless access point are you using? I was having a similar issue using Cisco enterprise  access points. My fix was to disable WMM Qos on the access points and everything is snappy.

Jchuchla, 2017-11-22:
I’m using the Ubiquiti Nanostation Loco M2.  I also have a UAP-AC-PRO inside that I use for testing.  They seem to perform pretty much the same.

--Jon Chuchla--

closed 2018-07-23 by forkineye