forkineye/ESPixelStick #96: GECE framerate /dropped frames is really bad on 3.0

Jchuchla, 2017-11-23:
Now that I've got my whole show up and running, the real sore spot is the gece snowflakes.  They have a very poor update rate.  They'll update mostly fine for several frames, and then skip several frames.  The sequence error counts on these pixelsticks are very high.  For Example:

Total Packets | 2980569
Sequence Errors | 444138
That means that at least 15% of the time, it's missing at least one frame.

Is there any workaround to get these looking smoother in the short term?  Each ESPixelStick is driving only 36 GECE bulbs.

penfold42, 2017-11-23:
Was it fine pre 3.0 ?

Is the wifi signal strength the same with both versions ?

Jchuchla, 2017-11-23:
Don’t know. Never used pixelsticks with gece before. But I can’t imagine this would be acceptable to anyone.

Shelby mentioned something about a change to how the geces ate bitbanged when he moved forward with the WiFi library. But it sounds like that was a long time ago.

--Jon Chuchla--

forkineye, 2017-11-23:
I can revert it to they way it was in 1.4 and test, but not until next week.

Jchuchla, 2017-11-23:
Yeah I realize timing is awful at the moment.
I’m trying to decide on whether to stick with it as is, switch out the bulbs to pixabulbs or change the controllers back to the komby stuff. The second option is probably the best. But I’m not sure if I have 180 pixabulbs laying around.

--Jon Chuchla--

forkineye, 2017-12-16:
For sake of keeping things up to date, a fix was implemented in the commit - https://github.com/forkineye/ESPixelStick/commit/4be708b723295f3b59481f529a309888a67d5491 and @Jchuchla has been testing it.  Thanks!

Jchuchla, 2017-12-16:
And for the sake of followup, it appears that this GECE code is much better than the 3.0rc3 version in my other snowflakes.
The only thing keeping me from rolling this out to the rest of the flakes is that the hostname stuff is broken in this build.

forkineye, 2017-12-16:
Yeah, the hostname stuff has to do with that build being compiled against lwip2. Digging into that in the core is on my to-do list.

forkineye, 2018-07-23:
@Jchuchla You recall for the hostname issue? I believe it had to do with dhcp reservations? Can't for the life of me think of it now.

Jchuchla, 2018-07-23:
I’m also struggling to remember the details.  You had built me a test for the GECE based on a newer IP stack or something like that.  This newer build did not let me connect to the device via hostname.  I had to get to it by IP.  When I looked in my DNS server, I saw no entry for that host.  It was pulling the DHCP address, but not registering the hostname in DNS.

But if I recall, this newer GECE build did show better performance.

FYI, I use MS DHCP and DNS on my network.  So I have pretty good tools to see what’s going on.

-- Jon Chuchla --

closed 2019-01-03 by forkineye