forkineye/ESPixelStick #97: Pwm crash on save ?

penfold42, 2017-11-23:
Have you seen my commitmxomments at 

https://github.com/forkineye/ESPixelStick/commit/f5b524bd3640f2448815768eac982d1bd60b141b

forkineye, 2017-11-23:
I have.  I won't be able to check them out til next week though. Thanks!

penfold42, 2017-12-02:
I still can’t get a crash to happen on my environment (other than by exceeding the max config file size)

forkineye, 2017-12-07:
I've been scratching my head on leaning the code a little bit as well.  The generate json is pretty large, brainstorming on making it more concise.

penfold42, 2017-12-07:
If we don’t care much for human readability we could make it tiny !

All the _enabled lines could reduced to a single line with a single number that’s a bit field (which is how it’s alsonstored internally)
Ditto for the _digital and _invert lines

Can you still reproduce the crash ? Even if we shrink the config, I’d like to determine why

forkineye, 2017-12-07:
I haven't had a chance to try it with your fix yet, may not be until after the season.

penfold42, 2018-03-15:
Have you found anything yet ?

penfold42, 2018-03-19:
I have good news - it’s failing for me now too. Not a crash, but the json parser hates the data

penfold42, 2018-03-19:
On closer inspection it looks like we’re not catering for ws messages that might be in multiple frames.

The example is ESPAsyncWebServer has additional logic for this

forkineye, 2018-03-19:
Haven't had time to work on the code lately, but I'm thinking about pulling the PWM stuff out and just keeping it in its own branch.

closed 2018-07-12 by forkineye