forkineye/ESPixelStick #100: Code nodemcu

vinc40530, 2017-12-04:
hello i wanted to put the code on my nodemcu but he found a problem on the udp including multicast variable can you please help me

penfold42, 2017-12-05:
I’m sure we can. We need some detail tho.

Do you have all the pre requisite library versions installed ?

We also need to see the errors...

vinc40530, 2017-12-05:
yes I have them all install and uninstall more times, I'm desperate, here is the exact error code
Arduino : 1.8.5 (Linux), Carte : "NodeMCU 1.0 (ESP-12E Module), 160 MHz, 115200, 4M (3M SPIFFS)"

/home/vinc/Arduino/libraries/ESPAsyncUDP-master/src/AsyncUDP.cpp: In member function 'bool AsyncUDP::listenMulticast(ip_addr_t*, uint16_t, uint8_t)':
/home/vinc/Arduino/libraries/ESPAsyncUDP-master/src/AsyncUDP.cpp:254:57: error: 'udp_set_multicast_netif_addr' was not declared in this scope
     udp_set_multicast_netif_addr(_pcb, multicast_if_addr);
                                                         ^
/home/vinc/Arduino/libraries/ESPAsyncUDP-master/src/AsyncUDP.cpp:258:36: error: 'udp_set_multicast_ttl' was not declared in this scope
     udp_set_multicast_ttl(_pcb, ttl);
                                    ^
exit status 1
Erreur de compilation pour la carte NodeMCU 1.0 (ESP-12E Module)

penfold42, 2017-12-05:
What version of esp8266 core do you have ?

2.40-rc2 or greater required

Required, not optional

vinc40530, 2017-12-05:
I don't know I plugged it without touching the tree

vinc40530, 2017-12-05:
here after the passage to rc2 it succeeded by cons I have no web page :( think I'm cursed

penfold42, 2017-12-05:
Did you also upload the spiffs which contains the web content ?

vinc40530, 2017-12-05:
yes as indicated

vinc40530, 2017-12-06:
I forgot the spot gulp, but how do you to launch it automatically, sorry I start in esp and code

vinc40530, 2017-12-06:
good evening everything works wonders a thousand thanks for your help against the script will know he can manage more than 680 leds because with the memory of nodemcu there is what to do

closed 2017-12-16 by forkineye