forkineye/ESPixelStick #101: V-3 DMX512 problem.

elektroservis, 2018-01-17:
V3.0 espixelstick mode DMX512

There are different stage lights at the workshop Robe brand. Address can be given. No problem when I connect to espixelstick at addresses 1-170. The stage robot does not receive a signal from the dmx to raise the adrese to 171 or higher.

When I try the same operation with v 2.0 dmx 512 mode espixelstick, all addresses are running smoothly.

elektroservis, 2018-02-01:
Hi,

When can you make changes about the subject?

djlazy, 2018-02-07:
Hi. I have same problem, DMX works to adress 170, and no data received after that. But in test mode, if i put to rainbow, fixtures react on later adresses also.

penfold42, 2018-02-17:
Does the view stream show all the channels with data ?

djlazy, 2018-02-22:
Stream testing show all channels, but not whole universe is on output, stops at channel 170.

elektroservis, 2018-02-26:
We are talking about the same problem,

When will it resolve? I'm looking forward to

forkineye, 2018-03-13:
@elektroservis @djlazy If you browse to /conf, what it is reporting for channel_count?  Also, what happens to the lights when you turn off test mode and while receiving DMX data? Do the ones past 170 stick in the last test mode color, go dark, or go into their default "no DMX data" mode?

djlazy, 2018-03-15:
Hi, this is part of data from conf:
"mqtt":{"enabled":false,"ip":"","port":1883,"user":"","password":"","topic":"diy/esps/4112db"},"e131":{"universe":1,"universe_limit":512,"channel_start":1,"channel_count":512,"multicast":true},"serial":{"type":0,"baudrate":250000}}

From 170 and further there is no data even in test mode.

leoguiders, 2018-03-28:
I'm having the same problem except that DMX channels up to 175 are working for me.

elektroservis, 2018-03-28:
@LeoGuiders the same problem continues. I have run 175 channels in my latest tests.

If the robot's address is 176, it is seen that it does not receive any signal.

leoguiders, 2018-03-28:
Interesting enough, I re-downloaded the source code and updated the Arduino core to 2.4.1, now everything is working as intended (512 DMX channels usable).

elektroservis, 2018-03-28:
@leoguiders How exactly did you do, please tell me in detail

elektroservis, 2018-03-28:
I tried it before you mentioned it. I did not get a result.

elektroservis, 2018-03-30:
@djlazy @leoguidersIs there a result you mentioned?

"Interesting enough, I re-downloaded the source code and updated the Arduino core to 2.4.1, now everything is working as intended (512 DMX channels usable)."

leoguiders, 2018-04-03:
@elektroservis I updated the Arduino core for ESP8266 WiFi chip to v2.4.1 (https://github.com/esp8266/Arduino/releases/tag/2.4.1), then I re-downloaded the v3.0 firmware release (https://github.com/forkineye/ESPixelStick/releases/tag/v3.0), built the file system (npm install, gulp) and uploaded everything via Arduino IDE.
In general, that was exactly what I did before, so I don't have a clue why it is working now.

jcollie, 2019-01-01:
I was able to get DMX addresses above 175 to work after reverting commit 8d09db7bbb10e7719d7c4e717af623e4bd97eff8 .

penfold42, 2019-01-01:
With what Arduino core version ?

jcollie, 2019-01-02:
> With what Arduino core version ?

2.4.2 (I use PlatformIO Core 4.0.0a1 as well).

DonMacarena, 2019-02-07:
Please tell me why in the DMX version only 272 channels are transmitted?   FW Version
3.1-dev Build Date Jan 16 2019

djlazy, 2019-08-13:
This problem is still present with esp-01 and esp-07.  in diagnostic view of stream there is data, but not on output. Works until channel 175

djlazy, 2019-08-13:
> @elektroservis @djlazy If you browse to /conf, what it is reporting for channel_count? 
Channel count is 512

Also, what happens to the lights when you turn off test mode and while receiving DMX data? Do the ones past 170 stick in the last test mode color, go dark, or go into their default "no DMX data" mode?

Ones after 175. goes to 0..

djlazy, 2019-08-13:
> I was able to get DMX addresses above 175 to work after reverting commit [8d09db7](https://github.com/forkineye/ESPixelStick/commit/8d09db7bbb10e7719d7c4e717af623e4bd97eff8) .

After doig this i have adresses 1 to 503 working

matajumotors, 2019-10-03:
I am getting 271 channel out only. on the ESPixelStick_TRAVIS-20190812180853. Could someone please compile verion that could be uploaded using UI or a release with the reverted commit 8d09db7? Thank you.

forkineye, 2020-01-11:
Sorry it took so long to take care of this, but it's now fixed in 4ea141711362ac32a059c3abed55c2261cf1d1c4.

closed 2020-01-11 by forkineye