forkineye/ESPixelStick #102: Delay problem. V-2

elektroservis, 2018-01-20:
Version 2

It takes about 5 seconds from power up to connect to the network, I want you to connect in 50 seconds. how to edit the code. which line?

penfold42, 2018-01-21:
From the 2.0 release tag, Espixelstick.ino line 166
There’s a 100ms delay there - make it longer should work.

But why ?

elektroservis, 2018-02-16:
Tahnk you,

Solved

penfold42, 2018-02-17:
Welcome, still curious - why ?

elektroservis, 2018-02-17:
Sorry.

After my router gives up energy, it starts after 50 seconds in average. When ESP and the router are energized at the same time, ESP will not find a router to connect to and will switch to AP mode. For this reason we needed to make such a change.

Thank you again for your interest.

also

V2 works wonderfully, no problem at all.

There is one more problem, you can get help, please. In the V3 DMX 512 mode only 170 channels are working. I suspect there is a place error in Serialdriver.h or SerialDriver.ccp. Could you help with this, please.

penfold42, 2018-02-17:
In that case, you’re better off editing :
#define CONNECT_TIMEOUT 15000   /* 15 seconds */

I don’t have serial so can’t help with that

elektroservis, 2018-02-17:
Thanks for your attention and quick reply.

Hopefully this error will be corrected soon. 

"V3 dmx 170 channel problem"

closed 2018-03-13 by forkineye