forkineye/ESPixelStick #103: Allow MQTT to set a longer string of values

Jchuchla, 2018-01-28:
I've been wanting to set multiple colors on a "string" via MQTT.  My particular application is my flood lights that have become year round landscape lighting.  I have more than one pixel flood on a chain.  I'd like to be able to set a pattern of multiple colors such as red and white for valentines day, or red white and blue for patriotic holidays.  My idea would be to allow for more values than just an RGB triplet in the RGB set topic.  So this got me looking into the current implementation in code.  I can see that it's not so simple to extend as currently written.  But it shouldn't be too hard to change it up so that it could parse the payload to collect the comma separated values into an array until the payload buffer is empty.  The array elements can then be counted and written as a pattern to the output buffer.
For example, if the payload is sent as 255,0,0,255,255,255,0,0,255 (red white blue) It would iterate the payload writing each of the 9 values to an array, and then find there's no more data.  At the same time, it would be incrementing a value counter NumValues. It would then memcopy those values to the output buffer incrementing by NumValues.  If the output buffer was set for 3 pixels in size, the output buffer would contain 255,0,0,255,255,255,0,0,255.255,0,0,255,255,255,0,0,255.255,0,0,255,255,255,0,0,255.

I've been seeing some steam behind MQTT and the ESPixelStick lately.  It's even prompted FPP to start implementing an MQTT daemon. Additional functionality here could be popular.

penfold42, 2018-01-28:
Would ASCII hex work for you?
80FF80 would be half saturated green.

No need for commas and each led would take 6 digits making the parser super simple.

Is there a length limit to mqtt messages ?

penfold42, 2018-01-28:
The limit of mqtt spec is more ram than an ESP8266 has!

This code has an effective limit right now of 256 bytes (which could be changed easily)

Jchuchla, 2018-01-28:
That would be functional, but I think the comma separated fields would be easier for a user to deal with. It’s much easier to build colors in your head in decimal rather than hex.

Jchuchla, 2018-01-28:
256 bytes would be more than sufficient for this.  I wouldn’t think the user would use such a feature for more than say 5 color triplets, so if using the comma separated ascii, that would be up to 60 bytes in the payload.

forkineye, 2018-07-26:
We're starting work on a configurable effects engine that should satisfy this. Discussion on it here - https://github.com/forkineye/ESPixelStick/issues/117.  Feed welcomed!

closed 2018-07-26 by forkineye