forkineye/ESPixelStick #105: ESP-07 and Web Saving Issues - Any way to hard code MQTT and Pixels as a Workaround?

robotica72, 2018-02-27:
I guess my first question is, does anyone know of a reason why the code may not work fully on an ESP-07?  I can flash it and it works great with the default settings but If I change any setting in the web GUI and save the settings, it stops responding and I cant reconnect until I reflash it with the base code again.

I have used the code on ESP-12's without any problems, but no luck on the ESP-07's.   

For my current project, It would be perfect if I could hardcode the MQTT settings and the pixel count.  I know the JSON file has the Universe settings and setting the Universe  in the JSON works, but I added sections for the pixel count and MQTT based on the source code - but none of the settings wrote when I flashed the code....

Anyone happen to have any ideas to try?  Either for the issues saving on the web GUI on the ESP-12 or a way to code in the MQTT and Pixel Count?

Thanks

penfold42, 2018-02-28:
Try erasing the flash chip ?

You can dump the config file from a running system by browsing to /conf
Note: from memory it’s missing the wifi password

robotica72, 2018-03-02:
I didnt try to erase the chip but I can - These were brand new chips and they flash and work fine, but just cant save the settings using the web page.   Since I can get to the web page, WiFi is good.

I wanted to get the output from the serial monitor to see if it helps out:

ESPixelStick v3.0 (Nov 20 2017)
*** Configuration File Format Error ***

Exception (28):
epc1=0x4021d264 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1370 end: 3fff15a0 offset: 01a0

>>>stack>>>
3fff1510:  3fff232c 3fff236c 3fff1540 402193a2  
3fff1520:  00000000 00000100 3fff04ac 4020c178  
3fff1530:  3ffe8dcc 0000000b 3fff04ac 4020b082  
3fff1540:  00000000 00000000 00000000 feefeffe  
3fff1550:  feefeffe feefeffe feefeffe feefeffe  
3fff1560:  feefeffe feefeffe feefeffe feefeffe  
3fff1570:  feefeffe feefeffe feefeffe 3fff056c  
3fff1580:  3fffdad0 00000000 3fff0565 40219d14  
3fff1590:  feefeffe feefeffe 3fff0580 4010070c  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

Same erro loops over and over after any changes and saving in the web gui but if I hook up an ESP-12, it programs without an issue?

forkineye, 2018-03-13:
I started receiving reports sometime last year and many people having issues with dodgy flash chips on 3rd party ESP-01 modules as well.  I haven't had any issues with the modules acquired from Ai-Thinker

Kamal-Sonani, 2018-03-13:
Why not just one code ?
With fix parameters like 
ssid "-----------"
Password"----------"
Universe (maximum as much as esp can handle)
Universe range "510"

No OTA changes ,that can reduce web save issues atleast those for who don't want many changes during setup.

robotica72, 2018-03-13:
Yeah, I saw some of the issues mentioning the chips - That was my first thought but they seem good with everything aside from the Web updates... So they might be good for what I need.

As Kamal-Sonani mentioned - My initial thought was to hardcode the information - but since I need to add the MQTT in there, I dont think the JSON edit is an option, I have no choice but to edit the Arduino code directly, unless Im missing something?

penfold42, 2018-03-13:
Why not edit the config.json ?

It’s easy if you’re building from source:
- run gulp if this is the first time
- simply edit data/config.json
- use sketch data upload

julienrat, 2019-09-03:
Hi !
Work like a charm with puya_V2 patch [https://github.com/esp8266/Arduino/issues/4061#issuecomment-368273656] and esp8266 2.4.2 core

Just go on esp8266 core folder (eg ~/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/)
Copy the patch inside this folder
And Patch ! `patch -b < puya_v2.patch `