forkineye/ESPixelStick #106: MQTT "on" being overridden by xlights/QLC+

swampdogmash, 2018-03-05:
Hi

Firstly, thanks for awesome firmware. Managed to follow all the instructions and get ESPixelStick v3.0 loaded onto a Wemos D1 mini and run a string of 50 WS2811 LEDs to it successfully. I've got QLC+ to talk to it using E131 and the same with Xlights. From what I understood when I enable MQTT and integrate the D1 into my homeautomation system I initially found that it DID override the E131 data, now however I am finding that it is not doing i, despite not having changed anything that I can think of.

When Xlights is open on my laptop and "light output" is disabled it still sends "off" to all the lights in the show. This seems to override the MQTT request to turn on - the LEDs turn on very briefly and then immediately turn off again.

Let me know if you need more info and I'll provide it, but I am just sending "MQTT /lights/set ON" and switch/set ON" along with a "rgb/set FFFFFF" and "brightness/set 30" one after the other.

forkineye, 2018-03-13:
If light output is disabled in xLights, it shouldn't be transmitting any E131 data.  It does look like MQTT is currently broken as its not being handled in the main loop.   Haven't had a chance to work on the code lately, but will try to get some fixes in this weekend.

swampdogmash, 2018-03-16:
Shelby...

I double checked the xlights output when it's "off" and as you correctly say there is no data. I had it in the situation where it was stopped displaying no sequence but output was on. Appreciate the correction. Did some more testing and do in fact appear to just be mqtt being overridden.. which is odd because I'm pretty sure at one point I WAS overriding it though I hadn't changed anything. Clearly that is next to impossible but as you have identified a potential bug I'll wait for a software revision and retest :)

swampdogmash, 2018-03-31:
Hows it going with this?.... you'll probably get a bunch more hits... just found this published on lighting channel... https://www.youtube.com/watch?v=BL0_sZFwl60&t=328s

Moonbase59, 2018-07-19:
Is this still an open issue? I can’t reproduce it on my latest build (3.1-dev + patches from my pull request).

Both xLights and QLC+ work correctly, MQTT overrides and only when I send a `"state":"OFF"` the current show continues flawlessly (both xLights and QLC+).