forkineye/ESPixelStick #111: lags on serial DMX512

Laurtib, 2018-06-13:
Hi,
I need to connect about 50 espixelsticks to a PI3 running Falcon Player.
I want to use a UNIFI wireless router; the PI3 will be wired connected to it.
The maximum distance from the router to the espixelsticks is about 25m (about 82ft) in open space.
The espixelsticks will run as DMX512 receivers.
It will be only one DMX512 univers, each espixelstick will output the same DMX512 univers; 
The espixelstick output will send dmx data to a PIC micro wich will decode the data and uses 3 channels.
This selection will be made from the espixelstick's web settings : 
for the first espixelstick : start channel = 1, channel count = 150,
for the second espixelstick : start channel = 2, channel count = 150, etc.
The first PIC micro will use the channels 1, 51 and 101, the second PIC micro will uses channels 2, 52, 102 ... 50, 100, 150.
I have setup the espixelsticks working in multicast mode.
**I have tested with 3 espixelsticks connected to the router and I have lags after 3-4min of correct working**.
Pls advise if my project is posible to make and how can I eliminate the lags.
thank you.

extremeskillz, 2018-07-11:
Multicast is bandwidth hungry in wireless. Keep your start changes but setup xlights to point their IP address in Unicast mode. The packets are smaller and probably will improve the lag time. 

Another question is what unifi AP are you using?

forkineye, 2018-07-23:
Multicast is ideal for this and what you want to be using. Which specific Unifi AP are you using?  Recommend the Nanostation LocoM2 and MAKE SURE that "Multicast Enhancements" are disabled or else it'll treat your multicast traffic like unicast traffic. Are you experiencing this just in DMX mode or pixel mode as well?

Laurtib, 2018-07-24:
Hi Shelby,
I'll use it in DMX mode
The espixelsticks will be placed on 3 discs, the bigest has 50m diameter 
This discs are placed at 8m high
I want to place the AP in the center of the disc, I do not know if  the Nanostation LocoM2 can cover radial space.

forkineye, 2018-07-24:
There should be adequate signal around it, but what AP were you using for testing when you experienced the lag?  Make sure you have no mobile devices, phones, tablets, laptops, etc... connected to the same AP when using multicast as it turns on power saving features in the wifi stack which will adversely affect multicast performance.

Laurtib, 2018-07-24:
Thank you for help,
I was using a EDIMAX AP.
Meantime I bought an UNIFI AP and I used 3 Espixelsticks (for now) in the network and they work without lags for more then 24 hours. I am happy about it.
The network I use is dedicated and there are no other devices connected to it.
The network will have only the 50 espixelsticks, one Raspberry PI 3 running Falcon Player and a router (the UNIFI AP has to be connected to a router in order to get an IP).
I am concerning about the other wifi networks that could work around mine; do you think my network activity could be affected ?

Moonbase59, 2018-07-25:
Now this is an interesting setup. By all means, keep us updated on how it worked out and what AP and settings you use!

My 2 cents on this:
* WiFi _can_ do "true" multicast, but often bad configuration or bad APs prevent using it.
* With ESP8266s, you can’ŧ use 5GHz, so it helps if not too much 2.4GHz traffic surrounds your setup.

By all means, try to:
* Use 802.11n mode (best bandwidth) and, if possible, disable 802.11g and at least 802.11b, because it slows down the whole WiFi network if the AP has to handle "slow" devices in between.
* You might want to sent a MAC filter in the AP so that others (like mobile phones) can’t even _try_ to log onto it and thus clog it.
* Some people recommend using fixed IPs in order to keep management traffic low. This isn’t strictly necessary but might help you keep an overview of "which ESPS is which".
* If anyhow possible, set "fixed transmission speed" in the AP to an appropriately high setting, because an AP will typically use the _lowest_ transmission speed if devices vary. Not many APs can do that, DD-WRT-based ones can. Example: 802.11g speeds are 6, 9, 12, 18, 24, 36, 48 and 54Mbps, you wouldn’t want it to negotiate 6Mbps, fix it to 54Mbps.
* If anyhow possible, try to disable multicast-to-unicast conversion in the AP. Sadly, some always do that.
* Use correct antenna orientation (i.e., AP & ESPixelStick both horizontal or vertical).  (A rod antenna like on an AP has best reception at 90° of where you point the rod, i.e. if it sticks up, reception would be best in a horizontal plane around it.)

Best of luck and let us know how it worked out!

forkineye, 2018-07-25:
@Laurtib RF noise from other devices will always be an issue, and I would let the AP settle on the rate.  Trying to force higher rates in a crowded spectrum can lead to other issues.  Even at 6Mbps, that's more than enough for a single universe and would be more reliable than trying to force a higher rate.  I've tried forcing rates on my setup in the past and it's always caused issues.

Laurtib, 2018-07-27:
@forkineye I'll let the default settings for the radio of the AP.
I have another concern : when I'll start the system, all the 50 ESPs will try to connect to the network.
I'd like to use the AP Fallback option and I wonder if it will appears some collisions or delays, so some of the ESPs not get connected in 30 seconds before goes to AP ?
A friend of mine suggested to insert a random delay before the ESP try to connect to the network.
I also think that I can start the ESPs in groups of 10 delayed with 5-10 seconds.
Pls advise.

forkineye, 2018-07-27:
There's a small random delay at startup (can be increased, first line of connectWifi()) to keep that from happening, so I don't think it would be an issue. I would recommend against using AP Fallback though as you could possibly get a stick to get stuck in AP mode if it doesn't connect the first try.

Laurtib, 2018-07-28:
ok, thanks.
In about 7-10 days I hope everything will be set and I'll let you know about how it works.

forkineye, 2019-01-03:
Closed due to inactivity.

closed 2019-01-03 by forkineye