forkineye/ESPixelStick #113: HomeAssistant: Doesn’t react on brightness, color

Moonbase59, 2018-07-18:
I just pulled the current master and am running ESPixelStick on a NodeMCU with a 60-pixel WS2812B strip and QLC+/xLights on Linux Mint 19, as well as with Home Assistant.

This version (3.1-dev) doesn’t react on the Home Assistant "color" and "brightness" commands because it wants the `"effect":"Solid"` in the MQTT JSON (which Home Assistant doesn’t send):

Here’s a short log from a mosquitto client subscribe (logging HA commands):
```
esps/1/set {"brightness": 99, "state": "ON"}
esps/1 {"state":"ON","color":{"r":127,"g":127,"b":127},"brightness":255,"effect":"Solid"}
esps/1/set {"color": {"r": 72, "g": 209, "b": 255}, "state": "ON"}
esps/1 {"state":"ON","color":{"r":127,"g":127,"b":127},"brightness":255,"effect":"Solid"}
```
As you can see, neither the _brightness_ nor the _color_ commands are respected.

If I now publish manual commands via `mosquitto_pub`, it works _if_ I include the `"effect":"Solid"`:
```
esps/1/set {"state":"ON","brightness":99,"effect":"Solid"}
esps/1 {"state":"ON","color":{"r":72,"g":209,"b":255},"brightness":99,"effect":"Solid"}
esps/1/set {"color": {"r": 72, "g": 209, "b": 255}, "state": "ON", "effect": "Solid"}
esps/1 {"state":"ON","color":{"r":72,"g":209,"b":255},"brightness":99,"effect":"Solid"}
```
Would you possibly change this not to require the `"effect":"Solid"` (shouldn’t be a big change, I’ll try myself later on) or do we really need to set up a HA template to include that?

In HA, I currently have:
```yaml
light:
  - platform: mqtt_json
    name: "ESPixelStick 1"
    state_topic: "esps/1"
    command_topic: "esps/1/set"
    brightness: true
    rgb: true
    effect: true
    effect_list:
      - Rainbow
      - Chase
```
Otherwise, apart from a few E1.31 sequence errors (currently 69 of 108958 packets) everything works quite well, even the switching over to MQTT (while a show is running) and back to DMX. Wonderful idea, by the way!

Thanks for all your great efforts to make E1.31 come near "normal" users!

Moonbase59, 2018-07-19:
Further testing with my code reveals that Home Assistant has no "Effect: OFF" – you’d have to switch the light off to reset it to "Solid".

Since we want to be able to actually _modify_ any effect’s color and brightness through a home automation system (instead of resetting), there is an easy workaround: Add the "Solid" effect to the light template in HA:

```yaml
light:
  - platform: mqtt_json
    name: "ESPixelStick 1"
    state_topic: "esps/1"
    command_topic: "esps/1/set"
    brightness: true
    rgb: true
    effect: true
    effect_list:
      - Solid
      - Rainbow
      - Chase
```

Now you can easily switch the light on and off, set color, brightness and effects, and even modify an effect’s color and brightness (if so supported—changing color on a Rainbow doesn’t make sense). :-)

I’d be happy if you’d include the code above in your master branch.

forkineye, 2018-07-19:
Can you do a PR with the changes?  If not, I can get it in this weekend, thanks for tracking this down!

Moonbase59, 2018-07-19:
Hee hee, I just tried! Hope I did everything right, it’s only my 2nd PR ever …

PR #115 (2018-07-20): Fix MQTT and new effects
* Fixed MQTT JSON handling, for better integration with home automation systems like Home Assistant.
* Added new effects "Blink" & "Flash".
* Updated Home Assistant example in README.md.
* Fixes #113, #114.
closed 2018-07-20 by forkineye