forkineye/ESPixelStick #116: Q: Control dumb 5050-type LED strips, run on H801?

Moonbase59, 2018-07-20:
I was just browsing the older code and recognized there must once have been some H801 support for "dumb" LED strips (PWM-based). Is that true and did it ever work? If yes, why was it removed?

I just got some H801 controllers and—after discovering ESPixelStick—now desperately look for something that I could flash on these to get MQTT _and_ DMX …

forkineye, 2018-07-23:
@penfold42 added PWM support and is supporting it in his fork.  It may come back in the future, but I need to refactor some things first so that other modes of operation can be supported in a cleanlier fashion.

closed 2018-07-23 by forkineye

Moonbase59, 2018-07-25:
Okay, thanks. I’d somehow love to see it back, like in "all the good stuff in _One_" ;-)