forkineye/ESPixelStick #123: ArduinoJson (and other libraries): Stable or newest?

Moonbase59, 2018-08-07:
The Arduino IDE keeps nagging be about updates for _ArduinoJson_. I currently kept the latest _stable_ (5.13.2), but they are at a _beta_ of 6.2.3-beta now.

To keep in sync and have comparable results: Should we use the _latest_ versions (including betas) or the _latest stable_ versions?

Or is there something like `package-lock.json` for the Arduino libraries where you could "nail" the versions to be used?

forkineye, 2018-08-07:
The issue is that not all of the libraries necessarily have defined stable releases. We could add hooks to the libraries as sub-modules but it gets messy with the way Arduino "auto-magically" builds.  Travis builds against the latest git pulls of everything so any glaring issues should pop up there.  When it gets closer to a binary release, I'll stick to the latest "stable" versions where I can.  Sometimes that's not possible though with things like the core where stable releases can be far between.

Moonbase59, 2018-08-09:
Hm, ok. So I try to keep the ArduinoIDE updated as far as possible but stick with _ArduinoJson_ 5.13.2 until the 6.2.x-beta is more stable (I’ve had problems with either 6.1.0 or 6.2.1 I think).

closed 2018-08-09 by Moonbase59