forkineye/ESPixelStick #128: Add persistent effects for devices with limited WiFi connectivity

forkineye, 2018-09-19:
This will consists of an auto-start option for effects in the test / effects tab along with a save button.  When checked and saved, the effect should persist and auto-start upon reboot regardless if a WiFi association is made. If enabled, the ESP8266 will not reboot if association reaches the timeout period.  Effect state should be saved in config.json.

extremeskillz, 2018-09-21:
Looks like a good option to me. Would be nice to have a prop in its own mode when not connected to FPP or Xlights for like a demo or independent effects to display.

penfold42, 2018-10-07:
I need this too for one of my use cases.
Home automation where I need the lights to default to “on x%” on a power cycle.

Ideally on ASAP before even trying to associate

LightUpMA, 2018-10-16:
I was going to request this feature; great minds think alike!!

penfold42, 2018-10-16:
@forkineye have you started this yet ?

If not I’ll make a start on it by doing the webUI and config parsing

penfold42, 2018-11-02:
All done in #136 except for the wifi startup changes.

penfold42, 2018-11-09:
What’s missing from ESPixelStick ?

Is it just more effects you want ?

mattncsu, 2018-11-09:
Nothing is missing, it does exactly what it says it should do. I just like the idea of some built in patterns so simple effects can be run without the need for FPP or xLights running on another machine. It would make it more one-size fits all if it supported streaming from a show source _and_ some built in effects.

penfold42, 2018-11-09:
Cool - there’s 8 effects there today.

Adding more is pretty easy.
First one that comes to mind is a 2 colour transition - a start colour and an end colour and have it smooth fade from one to the other

gjccat, 2018-11-09:
Can we have a save button to persist th the selected "effect" as opposed to having to change the default effect in the source.

penfold42, 2018-11-09:
Yes, that’s what my pull request #136 does

[image]

penfold42, 2019-01-06:
#136 is merged
 Close ?

forkineye, 2019-01-06:
Thanks @penfold42

closed 2019-01-06 by forkineye